RT-Thread/rt-thread #443: x86 can not be compiled in current version

wzyy2, 2015-03-26:
`include/rthw.h` had been changed before, which lead to compile problems  with ia32.

BernardXiong, 2015-03-26:
Show your compiling log. Thanks.

wzyy2, 2015-03-26:
drivers/board.c: 在函数‘rt_hw_board_init’中:
drivers/board.c:50:2: 警告：传递‘rt_hw_interrupt_install’的第 2 个参数时在不兼容的指针类型间转换 [默认启用]
In file included from drivers/board.c:17:0:
/Users/chen/GitHub/rt-thread/include/rthw.h:78:18: 附注：需要类型‘rt_isr_handler_t’，但实参的类型为‘void (*)(int)’
drivers/board.c:50:2: 错误：提供给函数‘rt_hw_interrupt_install’的实参太少
In file included from drivers/board.c:17:0:
/Users/chen/GitHub/rt-thread/include/rthw.h:78:18: 附注：在此声明

BernardXiong, 2015-03-28:
En, maybe the code is not updated to the new ISR handler mode.

Could you please provide me a x86 gcc-elf compiler link? Therefore, I can try it.

BernardXiong, 2015-03-28:
OK, I got the compiler. I will fix it soon.

BernardXiong, 2015-03-28:
... failed! It seems there is no newlib integrated in my used GCC compiler:
https://github.com/rm-hull/barebones-toolchain

Please provide me a i686-elf gcc.

wzyy2, 2015-03-29:
I compiled  it in ubuntu
    PREFIX="" and  DEVICE += ' -m32'

wzyy2, 2015-04-09:
another bug。。

    void rt_hw_interrupt_enable(rt_base_t level)
    {
    	    __asm__ __volatile__("pushl %0 ; popfl": :"g" (level):"memory", "cc");
    }

which should be 

    void rt_hw_interrupt_enable(rt_base_t level)
    {
            __asm__ __volatile__("pushl %0 ; popfl ; sti": :"g" (level):"memory", "cc");
    }

leezhenghui, 2015-09-11:
I actually see the issues mentioned as wzyy2 on my linuxmint 64bit:

scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build/bsp
CC build/bsp/applications/application.o
CC build/bsp/applications/startup.o
CC build/bsp/drivers/board.o
drivers/board.c: In function 'rt_hw_board_init':
drivers/board.c:50:2: warning: passing argument 2 of 'rt_hw_interrupt_install' from incompatible pointer type
  rt_hw_interrupt_install(INTTIMER0, rt_timer_handler, RT_NULL);
  ^
In file included from drivers/board.c:17:0:
/home/lizh/playground/uPlus/rt-thread/include/rthw.h:76:18: note: expected 'rt_isr_handler_t' but argument is of type 'void (*)(int)'
 rt_isr_handler_t rt_hw_interrupt_install(int              vector,
                  ^
drivers/board.c:50:2: error: too few arguments to function 'rt_hw_interrupt_install'
  rt_hw_interrupt_install(INTTIMER0, rt_timer_handler, RT_NULL);
  ^
In file included from drivers/board.c:17:0:
/home/lizh/playground/uPlus/rt-thread/include/rthw.h:76:18: note: declared here
 rt_isr_handler_t rt_hw_interrupt_install(int              vector,
                  ^
In file included from drivers/board.c:65:0:
/home/lizh/playground/uPlus/rt-thread/components/finsh/finsh.h: At top level:
/home/lizh/playground/uPlus/rt-thread/components/finsh/finsh.h:60:23: fatal error: sys/types.h: No such file or directory
 #include <sys/types.h>
                       ^
compilation terminated.
scons: *** [build/bsp/drivers/board.o] Error 1
scons: building terminated because of errors.

I am using the GCC compiler  "i686-elf-4.9.1-Linux-x86_64" as suggested  in the wiki.

leezhenghui, 2015-09-11:
I am studying the rt-thread code right now for my project, before I get a hardware board, I would like to try it on either bsp simulator or getting the x86 running on qemu. So the resolution to either #550 or this issue would helpful for me. thanks.

closed 2015-10-02 by BernardXiong