RT-Thread/rt-thread #455: SMP CPU Support in Raspberry Pi II (Cortex-A7)

BernardXiong, 2015-04-09:
It's planning to support SMP CPU in Raspberry Pi II, which has a Quad ARM Cortex-A7 CPU.

oska874, 2015-04-09:
������������������

BernardXiong, 2015-04-09:
@oska874 What are you saying?

wzyy2, 2015-04-09:
Wow!
I'm really interested in how os support SMP.

hduffddybz, 2015-04-10:
Hi! Recently I am doing the paper searching about the multiprocessor Real-Time Scheduling because I have to give speech about the parallel system and embedded system in our class.

I can list some classical papers talking about it though I did not read it carefully(you can see it from my dropbox(https://www.dropbox.com/sh/0jv49vif0cvyopj/AADbZv7uHjzypMHtnWX7HpAfa?dl=0):):

- The seminal paper for multiprocessor

On a Real-Time Scheduling Problem(Dhall and Liu 1978)

- Taxonomy of multiprocessor scheduling algorithms

A Categorization of Real-Time Multiprocessor Scheduling Problems and Algorithms(Carpenter et al. 2004)

- Phd paper from he University of North Carolina at Chapel Hill

SCHEDULING AND LOCKING IN MULTIPROCESSOR REAL-TIME OPERATING SYSTEMS(2011)

There are two fundamental methods in the multiprocessor Real-Time Scheduling:

- Partitioned Multiprocessor Real-Time Scheduling

Each task is assigned to a single processor so the processors are scheduled independently. It can extend the uniprocessor scheduling algorithm to the multiprocessor, for example from FP(fixed priority) to PFP.  The kernel should guarantee that  each task must be assigned to a processor that no processor is overload, a bin-packing-like problem can use to solve this problem.

- Global Multiprocessor Real-Time Scheduling

All available processors are scheduled using one scheduler. All ready tasks are stored in a single priority-ordered queue, the global scheduler selects highest priority tasks from this queue to execute.

Later I will do more paper reading work about it.

BernardXiong, 2015-04-12:
I prefer the SMP scheduling but each task can be binding to a specified Core with API.

hduffddybz, 2015-04-13:
What is the meaning of "SMP scheduling"? I think “SMP” are more suitable to describe the computer architecture rather than the scheduling algorithm.

BernardXiong, 2015-04-13:
A scheduler for SMP architecture, just for your mentioned: Global Multiprocessor Real-Time Scheduling.

bright-pan, 2015-04-14:
it`s cool...

BernardXiong, 2015-10-02:
Anybody can implement this feature in the near v2.1.0 version? Otherwise, I have to move it forward to v2.2.0 version.

closed 2022-06-16 by mysterywolf