RT-Thread/rt-thread #458: rt_tick_increase may need a check for the return of rt_thread_self()?

wzyy2, 2015-04-12:
void rt_tick_increase(void)
{
    struct rt_thread *thread;

    /* increase the global tick */
    ++ rt_tick;

        /* check time slice */
        thread = rt_thread_self();

    -- thread->remaining_tick;
    if (thread->remaining_tick == 0)
    {
        /* change to initialized tick */
        thread->remaining_tick = thread->init_tick;

        /* yield */
        rt_thread_yield();
    }

    /* check timer */
    rt_timer_check();
}

wzyy2, 2015-04-12:
if rt_thread_self() return NULL?

hduffddybz, 2015-04-12:
Hi! I think rt_thread_self() will never return NULL because if there are no any other higher tasks are ready the idle thread will run forever!

wzyy2, 2015-04-12:
But before scheduler start,rt_thread_self() is just return rt_current_thread, which is a uninitialized  pointer in bss section.

hduffddybz, 2015-04-12:
When start up the system, the interrupt will be closed you can see it in startup.c , but what I am not sure is that when will the interrupt be enabled? Maybe after rt_system_scheduler_start()?I have no hardware to do experiment now and I did not find rt_hw_interrupt_enable() in the source code!

wzyy2, 2015-04-12:
I check some bsp and I think there is no specific step to disable interrupt before scheduler starting...

I also have no hardware to test...

grissiom, 2015-04-12:
1, Most CPU boot in an interrupt-disabled state. Because not interrupted in the booting procedure is a common need.
2. BSP will call `rt_hw_interrupt_disable` in the first line of `main`. This is a de facto standard.

wzyy2, 2015-04-12:
well....I find `rt_hw_interrupt_disable` in the first line of `main`，but I think it's still a hidden trouble.

`bsp efm32` call `rt_hw_interrupt_enable` in `rtthread_startup`.
If I call `rt_malloc` out of `init_thread_entry`  will indirectly enable  interrupt.

BernardXiong, 2015-04-12:
Therefore, if the interrupt is enable and triggered before scheduler start up, it's a bug in the BSP. The global interrupt should be enable when system switch to the first task.

wzyy2, 2015-04-15:
I get it.. 
`rt_hw_interrupt_enable(base_t level)` is not `rt_hw_interrupt_enable()`.

closed 2015-04-15 by wzyy2