RT-Thread/rt-thread #492: 请问能否给个rtt2.1版本下，usb设备例子啊

baiheshu, 2015-06-03:
搞了好几天没头绪，不明白ep_pool如何初始化，能否给个使用说明之类的，提示一下流程

baiheshu, 2015-06-03:
struct udcd
{
    struct rt_device parent;
    const struct udcd_ops* ops;
    struct uendpoint ep0;
    struct ep_id* ep_pool;
	  struct rt_completion completion;
};
static rt_err_t rt_usbd_ep_assign(udevice_t device, uep_t ep)
{
    int i = 0;
    
    RT_ASSERT(device != RT_NULL);
    RT_ASSERT(device->dcd != RT_NULL);    
    RT_ASSERT(device->dcd->ep_pool != RT_NULL);        
    RT_ASSERT(ep != RT_NULL);
    RT_ASSERT(ep->ep_desc != RT_NULL);

    while(device->dcd->ep_pool[i].addr != 0xFF)
    {
        if(device->dcd->ep_pool[i].status == ID_UNASSIGNED && 
            ep->ep_desc->bmAttributes == device->dcd->ep_pool[i].type &&
            ep->ep_desc->bEndpointAddress == device->dcd->ep_pool[i].dir)
        {
            EP_ADDRESS(ep) |= device->dcd->ep_pool[i].addr;
            ep->id = &device->dcd->ep_pool[i];
            device->dcd->ep_pool[i].status = ID_ASSIGNED;

            RT_DEBUG_LOG(RT_DEBUG_USB, ("assigned %d\n", device->dcd->ep_pool[i].addr));  
            return RT_EOK;
        }
        
        i++;
    }
    
    return -RT_ERROR;
}

ep_pool是一个指针型的结构体，为什么函数中成了指针型的结构体数组了？

BernardXiong, 2015-07-05:
后面，USB device/host组件转到我这边来维护，希望能够早日给出相应的一些示例。

aric1987, 2017-02-19:
支持楼上，希望尽快发布一些USB例程出来。谢谢。

uestczyh222, 2017-11-15:
3.0.0已有可用usb bsp/stm32f10x_HAL 不兼容2.1的USB协议栈

closed 2018-04-27 by uestczyh222