RT-Thread/rt-thread #526: list_thread出现奇怪bug

tcz717, 2015-07-26:
当我使用finsh调用`list_thread()`时除了列出我已创建的线程，还会不断重复的列出`tshell`和`tidle`这两个线程，不会停止，如下面这个例子，请问如何解决。

```
finsh>>list_thread()
 thread  pri  status      sp     stack size max used   left tick  error
-------- ---- ------- ---------- ---------- ---------- ---------- ---
hmc5883  0x06 suspend 0x0000012c 0x00000300 0x0000012c 0x00000002 000
sonar    0x07 suspend 0x00000104 0x00000200 0x00000104 0x00000001 000
correct  0x0c suspend 0x00000124 0x00000300 0x00000124 0x00000001 000
control  0x03 suspend 0x00000164 0x00000300 0x0000022c 0x00000001 000
led      0x10 suspend 0x00000084 0x00000100 0x00000084 0x00000001 000
tshell   0x14 ready   0x0000019c 0x00000800 0x000001b0 0x00000005 000
tidle    0x1f ready   0x00000044 0x00000800 0x00000044 0x00000020 000
tshell   0x14 ready   0x0000019c 0x00000800 0x000001b0 0x00000005 000
tidle    0x1f ready   0x00000044 0x00000800 0x00000044 0x00000020 000
tshell   0x14 ready   0x0000019c 0x00000800 0x000001b0 0x00000005 000
tidle    0x1f ready   0x00000044 0x00000800 0x00000044 0x00000020 000
tshell   0x14 ready   0x0000019c 0x00000800 0x000001b0 0x00000005 000
tidle    0x1f ready   0x00000044 0x00000800 0x00000044 0x00000020 000
tshell   0x14 ready   0x0000019c 0x00000800 0x000001b0 0x00000005 000
tidle    0x1f ready   0x00000044 0x00000800 0x00000044 0x00000020 000
tshell   0x14 ready   0x0000019c 0x00000800 0x000001b0 0x00000005 000
tidle    0x1f ready   0x00000044 0x00000800 0x00000044 0x00000020 000
tshell   0x14 ready   0x0000019c 0x00000800 0x000001b0 0x00000005 000
tidle    0x1f ready   0x00000044 0x00000800 0x00000044 0x00000020 000
tshell   0x14 ready   0x0000019c 0x00000800 0x000001b0 0x00000005 000
tidle    0x1f ready   0x00000044 0x00000800 0x00000044 0x00000020 000
tshell   0x14 ready   0x0000019c 0x00000800 0x000001b0 0x00000005 000
tidle    0x1f ready   0x00000044 0x00000800 0x00000044 0x00000020 000
tshell   0x14 ready   0x0000019c 0x00000800 0x000001b0 0x00000005 000
tidle    0x1f ready   0x00000044 0x00000800 0x00000044 0x00000020 000
..........
```

BernardXiong, 2015-07-26:
你的finsh线程被多次初始化了。

tcz717, 2015-07-26:
非常谢谢，已解决。我这`startup.c`和`components.c`两个文件的确都初始化finsh了。

closed 2015-07-26 by tcz717