RT-Thread/rt-thread #533: 关于libc和scanf

tcz717, 2015-08-06:
在 `/components/libc/armlibc/stubs.c`的
```
int _sys_read(FILEHANDLE fh, unsigned char *buf, unsigned len, int mode)
{
#ifdef RT_USING_DFS
    int size;
#endif

    if (fh == STDIN)
    {
        /* TODO */
        return 0;
    }

    if ((fh == STDOUT) || (fh == STDERR))
        return -1;

#ifndef RT_USING_DFS
    return 0;
#else
    size = read(fh - STDERR - 1, buf, len);
    if (size >= 0)
        return len - size;
    else
        return -1;
#endif
}
```
关于STDIN的实现还处于TODO的阶段，请问有什么比较好的方法实现从而可以使用scanf，例如采用linux那种tty映射的方式么？

BernardXiong, 2015-08-13:
这个关键在于当前从device read数据和传统的有些不兼容，例如说，getc --> rt_device_read，例如是从串口中获得数据，然后如果是INT_RX方式的串口，rt_device_read就直接返回了。

这块可能需要进行一定的调整，需要让_sys_read和console->rx_ind挂钩起来。

mysterywolf, 2021-02-28:
经测试，scanf现在可以使用了，不过同是使用shell和scanf的话，shell线程会抢scanf输入的数据。

closed 2021-02-28 by mysterywolf