RT-Thread/rt-thread #535: 建议将某些参数定义成枚举类型

heyuanjie87, 2015-08-11:
rt_mutex_t rt_mutex_create(const char *name, rt_uint8_t flag) 参数中的flag取值比较固定 但有时候又不记得是啥，每次用起来都会脚得迷茫 紧张 忐忑。

BernardXiong, 2015-08-11:
这个确实是的，不过枚举类型也是最令人沮丧的类型，编译器分配出来的长度是不固定的（而C++的枚举类型判断也是噩梦，导致定义的类型和类型进行 | 运算时，编译器直接就抛出警告或编译失败）。或者有更好的方式？

XilongPei, 2015-08-12:
typedef

heyuanjie87, 2015-08-12:
在函数内部强制类型转换后再运算不知道会不会报错

BernardXiong, 2015-08-13:
typedef和enum类型没关系

BernardXiong, 2015-08-13:
@heyuanjie87 强制类型转换？怎么转换？C++的吗？

XilongPei, 2015-08-13:
Elastos操作系统源码中，我们一般是，比如： 

/**

*  关于XXX取值范围的描述

  */

typedef XXX unsigned long；

 

XXX实际上是一个标识，因为编译器是没办法替你做这类检查的，但是，具体写程序时，遇到XXX类型的，知道，它是一个有特殊要求的类型，实际上它就是一个无符号32位整型值。

BernardXiong, 2015-08-13:
@XilongPei  原来是裴老师

是这样的， @heyuanjie87 提到的是，希望能够直接使用enum类型，这样就方便在编写代码时直接（在IDE中）使用相关的带名称的值。而enum类型主要的问题在于enum类型的大小不固定，当和C++混用时，类似 ENUM_AA_ | ENUM_BB_ 这样的混用会被C++编译器当警告或错误来处理，所以目前RT-Thread编码尽量少的使用enum类型。

而类似你上面提到的，用typedef来定义个32位无符号类型，实际上和heyuanjie提到的无关，并不能够解决这样的问题。

heyuanjie87, 2015-08-13:
测试了一下 的确会报错
typedef enum
{
    FLAG1 = 0x01,
    FLAG2 = 0x02,
}flag_t;

void func(flag_t f)
{ }

int main()
{
	func((flag_t)(FLAG1 | FLAG2));需要位运算时只有弄成这样
      不过貌似在大多数传flag的时候都很少会有 | 操作。如果需要，可不可以再定义一个名称的值，是它们运算过后的值。比如 ENUM_R = 1，ENUM_W = 2，ENUM_RW = 3这样的方式。
}

closed 2015-09-17 by BernardXiong