RT-Thread/rt-thread #537: 关于主分支下/BSP/STM32F0工程使用SCONS编译后仍然编译出错

lishidetiankong, 2015-08-18:
按照Bernard Xiong的提示，使用SCONS将工程重新编译了一遍但是仍然提示有错误，还是无法找到components/components.c文件，将工程列表中的components/components.c重新定向为src/components.c这个文件，仍然无法编译，提示如下错误:
.\build\template.axf: Warning: L6320W: Ignoring --keep command. Cannot find argument '__fsym_*'.
.\build\template.axf: Warning: L6320W: Ignoring --keep command. Cannot find argument '__vsym_*'.
.\build\template.axf: Error: L6218E: Undefined symbol msh_auto_complete (referred from shell.o).
.\build\template.axf: Error: L6218E: Undefined symbol msh_exec (referred from shell.o).
.\build\template.axf: Error: L6218E: Undefined symbol msh_is_used (referred from shell.o).

heyuanjie87, 2015-08-18:
试试 scons --target=mdk4,如果是gcc的话我就没试过了

BernardXiong, 2015-08-19:
@lishidetiankong 你应该并未使用scons来生成工程。

针对MDK4/5，我已经从新用scons生成工程了，可以更新代码下来试试。

closed 2015-09-17 by BernardXiong