RT-Thread/rt-thread #541: 最近玩canfestival需要微秒级的定时器，弄了个硬件定时器驱动框架 有需要合并进来吗

heyuanjie87, 2015-08-29:


BernardXiong, 2015-08-30:
有，但是你得给个好的，单独套套rt_device的接口没太大意思。

BernardXiong, 2015-09-04:
我觉得你也可以给出些，目前你了解到的一些使用方法，例如posix timer。

BernardXiong, 2015-09-18:
@heyuanjie87 
The following are the POSIX timer interface, for your refer:
timer_create - http://pubs.opengroup.org/onlinepubs/9699919799/functions/timer_create.html
timer_delete - http://pubs.opengroup.org/onlinepubs/9699919799/functions/timer_delete.html

clock_gettime/clock_settime
http://pubs.opengroup.org/onlinepubs/9699919799/functions/timer_getoverrun.html#

BernardXiong, 2015-10-02:
More information from opengroup.org for high resolution timer:
http://pubs.opengroup.org/onlinepubs/000095399/functions/clock_nanosleep.html
http://pubs.opengroup.org/onlinepubs/009604599/functions/nanosleep.html

Please adjust your implementation.

gbcwbz, 2022-05-15:
- 建议实现 clock_nanosleep clock_gettime 像 [Xenomai](https://xenomai.org/documentation/xenomai-3/html/xeno3prm/group__cobalt__api__time.html) 这种 
 - 在 EtherCAT 中不仅要实现相对定时，而且要实现绝对定时（为了将 EtherCAT 主站的循环时间和从站的循环时间同步），clock_nanosleep 就可以选择在将来的绝对时间唤醒当前线程

closed 2022-06-12 by mysterywolf