RT-Thread/rt-thread #550: Simulator compile failed on Linuxmint 64bit

leezhenghui, 2015-09-11:
I think I need to ask the question of whether simulator support 64bit linux host environment before jump into the compile issue, if yes, pls go ahead..

I keep seeing the error message below during compile simulator bsp:

CC build/kernel/src/scheduler.o
CC build/kernel/src/thread.o
CC build/kernel/src/timer.o
LINK rtthread
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libm.so when searching for -lm
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libm.a when searching for -lm
/usr/bin/ld: skipping incompatible /usr/lib/x86_64-linux-gnu/libm.so when searching for -lm
/usr/bin/ld: skipping incompatible /usr/lib/x86_64-linux-gnu/libm.a when searching for -lm
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-linux-gnu/4.8/libgcc.a when searching for -lgcc
/usr/bin/ld: cannot find -lgcc
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-linux-gnu/4.8/libgcc_s.so when searching for -lgcc_s
/usr/bin/ld: cannot find -lgcc_s
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libpthread.so when searching for -lpthread
/usr/bin/ld: skipping incompatible /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../x86_64-linux-gnu/libpthread.a when searching for -lpthread
/usr/bin/ld: skipping incompatible /usr/lib/x86_64-linux-gnu/libpthread.so when searching for -lpthread
/usr/bin/ld: skipping incompatible /usr/lib/x86_64-linux-gnu/libpthread.a when searching for -lpthread
collect2: error: ld returned 1 exit status
scons: *** [rtthread] Error 1
scons: building terminated because of errors. 

I guess this is because I am using 64bit linux, any comments/suggestion to compile and run the simulator is appreciated.

BernardXiong, 2015-09-12:
It seem you used a wrong gcc toolchain. Please use x86-elf-gcc toolchain.

leezhenghui, 2015-09-12:
Hi, Bernard

I updated my gcc toolchain to "x86_64-elf-4.9.1-Linux-x86_64", got a similar issue(a little different than the previous, but still complain for the option related to -pthread),  below is the error message I got during compilation:

scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build/bsp
LINK rtthread
x86_64-elf-gcc: error: unrecognized command line option '-pthread'
scons: *** [rtthread] Error 1
scons: building terminated because of errors.

Any thoughts about this?

BernardXiong, 2015-09-13:
Hi Lee, 
Could you please provide a toolchain for me? Therefore, I can try some things.

leezhenghui, 2015-09-14:
Hi, Bernard

I am using the toolchain download from [1], my development environment is linuxmint 64bit, it essentially based on ubuntu 14.04 64bit. So I suppose you can reproduce the issue on ubuntu 14.04 64bit. If you want to reproduce the issue locally, cloud9 is quick way to do this, as cloud-9 supplies dev env is based ubuntu 14.04 64bit

[1] http://newos.org/toolchains/x86_64-elf-4.9.1-Linux-x86_64.tar.xz
[2] http://wiki.osdev.org/GCC_Cross-Compiler

BernardXiong, 2015-09-14:
Is x86_64-elf used for 64bit ELF?

leezhenghui, 2015-09-14:
Hi, Bernard

Yes, I think so, and I also double checked with those compiled object file, these object files also tell us it is ELF 64-bit, e.g:

clock.o: ELF 64-bit LSB  relocatable, x86-64, version 1 (SYSV), not stripped

Are you implying that I am using a wrong gcc-compiler toolchains? if so, appreciated that you point me to a specific toolchains you suggested to use in this case. 

BTW, I moved to this toolchain following your early suggestion of "x86-elf-gcc toolchain", before that, I was using the toolchain shipped by ubuntu. Let me know if I misunderstand your points there.

BernardXiong, 2015-09-15:
@leezhenghui 

I think it may be this version:
http://newos.org/toolchains/i686-elf-4.9.1-Linux-x86_64.tar.xz

I will try it in c9 later.

BernardXiong, 2015-09-15:
@leezhenghui 
I have fixed the x86 bsp compiling issue, Please update your code. However, I can not confirm whether it can run successfully.

leezhenghui, 2015-09-16:
Close the issue as the x86 bsp can be compiled successfully. Thanks Bernard

closed 2015-09-16 by leezhenghui