RT-Thread/rt-thread #552: 移植RTT 2.0.1 bsp下的lpc178x例程到我的lpc1778板子上，现在串口可以输出版本信息，但是finsh中一直没有有效信息输出

xtaens, 2015-09-18:
如题，我把2.0.1bsp下的lpc178x例程移植到我的lpc1778板子上，现在串口可以输出log信息，但是finsh中一直没有输出信息：数据如下

 \ | /
- RT -     Thread Operating System
 / | \     2.0.1 build Sep 16 2015
 2006 - 2015 Copyright by rt-thread team
finsh>>
finsh>>
finsh>>
finsh>>
finsh>>
finsh>>
finsh>>
finsh>>
finsh>>
finsh>>
finsh>>
finsh>>

请问这是哪里的问题呢？我已经使能了组件了，但是就是不行。

下面是我的配置文件：
/* RT-Thread config file */
#ifndef __RTTHREAD_CFG_H__
#define __RTTHREAD_CFG_H__

// <RDTConfigurator URL="http://www.rt-thread.com/eclipse">

// <integer name="RT_NAME_MAX" description="Maximal size of kernel object name length" default="6" />
#define RT_NAME_MAX	6
// <integer name="RT_ALIGN_SIZE" description="Alignment size for CPU architecture data access" default="4" />
#define RT_ALIGN_SIZE	4
// <integer name="RT_THREAD_PRIORITY_MAX" description="Maximal level of thread priority" default="32">
// <item description="8">8</item>
// <item description="32">32</item>
// <item description="256">256</item>
// </integer>
#define RT_THREAD_PRIORITY_MAX	32
// <integer name="RT_TICK_PER_SECOND" description="OS tick per second" default="100" />
#define RT_TICK_PER_SECOND	100
// <section name="RT_DEBUG" description="Kernel Debug Configuration" default="true" >
#define RT_DEBUG
// <bool name="RT_THREAD_DEBUG" description="Thread debug enable" default="false" />
// #define RT_THREAD_DEBUG
// <bool name="RT_USING_OVERFLOW_CHECK" description="Thread stack over flow detect" default="true" />
#define RT_USING_OVERFLOW_CHECK
// </section>

// <bool name="RT_USING_HOOK" description="Using hook functions" default="true" />
#define RT_USING_HOOK
// <section name="RT_USING_TIMER_SOFT" description="Using software timer which will start a thread to handle soft-timer" default="true" >
// #define RT_USING_TIMER_SOFT
// <integer name="RT_TIMER_THREAD_PRIO" description="The priority level of timer thread" default="4" />
#define RT_TIMER_THREAD_PRIO	4
// <integer name="RT_TIMER_THREAD_STACK_SIZE" description="The stack size of timer thread" default="512" />
#define RT_TIMER_THREAD_STACK_SIZE	512
// <integer name="RT_TIMER_TICK_PER_SECOND" description="The soft-timer tick per second" default="10" />
#define RT_TIMER_TICK_PER_SECOND	10
// </section>

// <section name="IPC" description="Inter-Thread communication" default="always" >
// <bool name="RT_USING_SEMAPHORE" description="Using semaphore in the system" default="true" />
#define RT_USING_SEMAPHORE
// <bool name="RT_USING_MUTEX" description="Using mutex in the system" default="true" />
#define RT_USING_MUTEX
// <bool name="RT_USING_EVENT" description="Using event group in the system" default="true" />
#define RT_USING_EVENT
// <bool name="RT_USING_MAILBOX" description="Using mailbox in the system" default="true" />
#define RT_USING_MAILBOX
// <bool name="RT_USING_MESSAGEQUEUE" description="Using message queue in the system" default="true" />
[... 138 lines cut ...]
// <bool name="RTGUI_USING_SMALL_SIZE" description="use small size in RT-Thread/GUI" default="true" />
#define RTGUI_USING_SMALL_SIZE
// <bool name="RTGUI_USING_FONT16" description="support 16 weight font" default="true" />
#define RTGUI_USING_FONT16
// <bool name="RTGUI_USING_FONT12" description="support 12 weight font" default="true" />
// #define RTGUI_USING_FONT12
// <bool name="RTGUI_USING_FONTHZ" description="support Chinese font" default="true" />
#define RTGUI_USING_FONTHZ
// <integer name="RTGUI_DEFAULT_FONT_SIZE" description="default font size in RT-Thread/GUI" default="16" />
#define RTGUI_DEFAULT_FONT_SIZE	16
// <bool name="RTGUI_USING_DFS_FILERW" description="use RT-Thread/DFS as file interface" default="true" />
// #define RTGUI_USING_DFS_FILERW
// <bool name="RTGUI_USING_HZ_BMP" description="use Chinese font bitmap engine" default="true" />
#define RTGUI_USING_HZ_BMP
// <bool name="RTGUI_USING_HZ_FILE" description="use font file as Chinese font" default="false" />
// #define RTGUI_USING_HZ_FILE
// <bool name="RTGUI_USING_MOUSE_CURSOR" description="use mouse cursor" default="false" />
// #define RTGUI_USING_MOUSE_CURSOR
// </section>

// </RDTConfigurator>

#endif

其他uart等文件都是bsp下的，该例程好像没有用组件功能，但是我按照stm32的做法，加入组件后，这个finsh不工作。。。

BernardXiong, 2015-09-18:
默认的finsh是工作的，但是加入其他组件后就不工作了？

xtaens, 2015-09-18:
2.0.1的lpc178x bsp 中finsh默认是不开启的，rtconfig.h中宏RT_USING_COMPONENTS_INIT都没屏蔽了。我已经在stm32f103中成功跑通了rtt，现在项目上想用rtt，但是芯片是lpc1778，所以就把bsp中的lpc178x按照stm32中的配置及uart文件改了一下，现在在lpc1778中finsh不输出任何有用的信息，即finsh提示符可以正常输出，按enter键finsh也可以正常再输出一次，按table键和按enter效果一样，不管输入什么命令都显示：Unknown symbol

xtaens, 2015-09-18:
\ | /
- RT -     Thread Operating System
 / | \     2.0.1 build Sep 16 2015
 2006 - 2015 Copyright by rt-thread team
finsh>>
finsh>>
finsh>>
finsh>>list()
Null node
finsh>>list
Unknown symbol
finsh>>version()
Null node
finsh>>
finsh>>
finsh>>
finsh>>

finsh组件及内核代码我在stm32上测试都没有问题，变化的就是uart、rtconfig.h和board.c文件不同

xtaens, 2015-09-18:
还没有加入其它组件，只测试了finsh。。。

这个评论框怎么把调试信息贴上后就全部挤到一样了呢？最后特意加一行换行，上面才显示的比较好看一些。

xtaens, 2015-09-18:
问题已经解决，由于我用的是IAR ARM ，和keil有些区别。

解决方法：在icf文件中加入下面的语句即可：

keep { section FSymTab };
keep { section VSymTab };
keep { section .rti_fn* };

BernardXiong, 2015-09-18:
是的，RT-Thread本身可以支持IAR，支持IAR相应的汇编等等。但是真正的在IAR上使用的比较少。

so，如果有什么更改，欢迎发pull request。

xtaens, 2015-09-18:
大侠好，lpc178x例程中没有IAR的工程，所以我的这个改动也没法加入到工程。

话说那个scons环境搭建太费劲了，而且生成的IAR工程也和我想要的不同，故一直没有建立scons环境。。。我都是自己建立的IAR工程。

closed 2015-10-02 by BernardXiong