RT-Thread/rt-thread #562: STM32F4 BSD的rtconfig问题

JGINGIT, 2015-10-10:
在BSD为STM32F4的```rtconfig.h```文件中，未有定义```#define CHECKSUM_GEN_ICMP 				0```导致默认```CHECKSUM_GEN_ICMP```为**1**, 又lwip默认使用的硬件校验，所以导致ICMP校验总是出错，表现为ping响应超时，定义了```#define CHECKSUM_GEN_ICMP 				0```之后问题解决。

BernardXiong, 2015-10-14:
@aozima 确认下

pangweishen, 2015-12-16:
应该是stm32f4xx_eth.c默认开启了CHECKSUM_BY_HARDWARE，而rtconfig.h中未定义CHECKSUM_GEN_ICMP，lwip默认使用软件校验和算法，而MAC采用硬件校验和，两者都生效了。

xools, 2017-04-22:
#define CHECKSUM_CHECK_ICMP             0
#define CHECKSUM_GEN_ICMP                  0
需要在rtconfig.h里定义上面两个宏，否则会有问题。主要是Lwip生成了ICMP的校验和之后，发送时如果eth驱动也是默认全部生成校验和，则反而会清零校验和位置。
我发现好几个bsp都没有define上面两个宏，可能会有问题。但我也只在stm32f407平台上测试过。

BernardXiong, 2017-04-23:
一般会考虑默认不开启硬件checksum的特性，由最终用户确定是否开启。

mysterywolf, 2021-04-17:
issue关闭了，目前标准库已经淘汰掉了。

closed 2021-04-17 by mysterywolf