RT-Thread/rt-thread #568: LWIP_ASSERT("pbuf_free: p->ref > 0", p->ref > 0);断言失败

Hotwill, 2015-10-14:
只要启用lwip，终端就会不断打印出：
pbuf_free: p->ref > 0
Assertion: 650 in ..\..\components\net\lwip-head\src\core\pbuf.c, thread tcpip
最终定位到pbuf.c中的LWIP_ASSERT("pbuf_free: p->ref > 0", p->ref > 0);断言失败导致的。
这应该是一个已经释放的pbuf被再次释放引起的。请修复这一问题。

BernardXiong, 2015-10-14:
请使用lwip-1.4.1的版本，这个是默认的lwIP版本。

Hotwill, 2015-10-14:
默认的不就是lwip-1.4.1的吗？我使用的就是这个版本

BernardXiong, 2015-10-14:
Assertion: 650 in ....\components\net\lwip-head\src\core\pbuf.c, thread tcpip

你的log信息出卖了你。

Hotwill, 2015-10-14:
哦，明白了

Hotwill, 2015-10-14:
lwip-head是支持IPv4/v6双栈的lwip，这个是在1.4.1的基础上开发的，我要用到ipv6，所以必须用这个。

BernardXiong, 2015-10-14:
IPv4/v6双栈的lwIP都还未正式发布呢

Hotwill, 2015-10-15:
我从github上下载了最新的rt-thread，什么都没有修改，也没有启用ipv6。只写了一个简单的服务端测试程序。在传输的时候还是会出现这样的问题：
[image]

下面是我的主要代码：
void tcp_server_thread_entry(void *parameter)
{
	int port = 3001;
    int sockfd, clientfd, bytes_received;
    struct sockaddr_in my_addr, their_addr;
	socklen_t sin_size;
	char buf[1024];
    
    sockfd = socket(AF_INET, SOCK_STREAM, 0);
	
	my_addr.sin_family = AF_INET;
	my_addr.sin_port = htons(port);
	my_addr.sin_addr.s_addr = INADDR_ANY;
	rt_memset(&my_addr.sin_zero, 0, 8);
	
	if (bind(sockfd, (struct sockaddr *)&my_addr, sizeof(my_addr)) == -1)
	{
		rt_kprintf("Bind local port %d failed!\n", port);
		return;
	}
	
	if (listen(sockfd, 5) == -1)
	{
		rt_kprintf("Listen failed!\n");
		return;
	}
	
    while (1)
	{
        clientfd = accept(sockfd, (struct sockaddr *)&their_addr, &sin_size);
		rt_kprintf("Get connection from %s\n", inet_ntoa(their_addr.sin_addr));

        while (1)
		{
            bytes_received = recv(clientfd, buf, sizeof(buf), 0);
            if (bytes_received <= 0)
			{
				rt_kprintf("Disconnect with client!\n");
                closesocket(clientfd);
                break;
            }
			
			rt_kprintf("Recv %d!\n", bytes_received);
        }
    }
}

BernardXiong, 2015-11-17:
Hi, 
当前这个问题是什么状态？如果是 lwip_head，是否可以使用最新的lwip版本试试？

imlvye, 2016-04-14:
遇到同样的问题，同样时LWIP 1.4.1
finsh>>a connection close
try accept
Got a connection from (192.168.1.101, 2388)
pbuf_free: p->ref > 0
Assertion: 650 in /home/xxxxxxxx/rtthread/components/net/lwip-1.4.1/src/core/pbuf.c, thread tcpip
(0) assertion failed at function:sys_arch_assert, line number:592
请问解决方法了吗？

Hotwill, 2016-04-18:
还没有解决

cmheia, 2017-09-12:
这个问题有什么进展吗？我这的程序在更换网卡驱动后出现了这个断言。

tranthudo, 2018-11-23:
I"m getting the same poblem.

24540477zxc, 2019-12-24:
the same problem at 4.0.2

tamerxu, 2021-10-16:
https://www.cnblogs.com/woshixiaojingzai/p/13958572.html   我认为这篇文章有些参考价值，各位可以分析看看，是否能解决，或者有没有更好的方法

LMRD-Mars, 2022-02-04:
有进展吗

xiangxistu, 2022-03-25:
我们这边的一般的 eth 驱动里，pbuf 一般都是 pbuf pool 的形式。 pbuf pool 数量也是在 lwipopt.h 中指定的内存池大小。
重复释放问题确实可能存在，但是可能是工程里的对 pbuf 的处理时具体的使用方法有关。

这个问题的复现，我这边还没有必现的工程，能否提供一下？

SunJun8, 2022-03-26:
@xiangxistu 
用st的话大部分应该是直接在eth用的sram上直接构造pbuf，容易出现同一块内存被使用两次

SanTaiZi-NeZha, 2023-04-20:
> 

不行，试了，加了锁还是一样