RT-Thread/rt-thread #572: bsp/stm32f10x中的canapp.c编译无法通过(MDK4 Project)

Xiaoyu0601-Wang, 2015-10-21:
我把最外围的这个定义注释掉之后，编译会有很多错误。
"#ifdef RT_USING_CAN
#endif"
有一些没有定义的结构体，例如stm32_hw_pin_userdata.
rt_can_filter_item filter1item[4] 里还有很多错误
另外我有些混乱。有两个文件bxcan.c, canapp.c, 这两个文件的关系是什么?
工程里并没有用到bxcan.c。

GrandHsu, 2015-10-28:
同问。
感觉文件可能不完整！

AubrCool, 2015-10-28:
https://github.com/AubrCool/rt-thread/tree/fix-stm32f10x.compile.err
https://github.com/AubrCool/rt-thread/tree/fix-can.hdr.api
先试试这两个,device 选的是stm32f105,其它的片子,中断号可能要改一下

Xiaoyu0601-Wang, 2015-10-28:
@AubrCool 好的，谢谢。

Xiaoyu0601-Wang, 2015-10-29:
@AubrCool 我用的KEIL MDK，有个不太明白的问题。
struct rt_can_filter_config filter1 =
{
    .count = 4,
    .actived = 1,
    .items = filter1item,
};
所有这类的结构体赋值，我都编译不过，都是expected an expression。
是编译器问题？还是头文件的问题？

Xiaoyu0601-Wang, 2015-10-29:
@GrandHsu 需要在Options/C/C++里把C99 Mode 选上，结构体编译才不会出问题。

zhitian, 2017-02-19:
把编译器配置成 --c99 就可以编译通过了

closed 2021-02-06 by mysterywolf