RT-Thread/rt-thread #606: 编译zynq7000 失败

oska874, 2016-01-11:
按照readme.md 的说法，`cd` 到`bsp/zynq7000`，然后执行命令`scons`，系统开始build，但是最后报错
```
In file included from /workspace/github/rtos/rt-thread/include/components.h:72:0,
                 from applications/application.c:23:
/workspace/github/rtos/rt-thread/components/pthreads/pthread.h:132:33: error: conflicting type qualifiers for 'pthread_spinlock_t'
In file included from /usr/lib/gcc-cross/arm-linux-gnueabi/4.7/../../../../arm-linux-gnueabi/include/sys/types.h:270:0,
                 from /workspace/github/rtos/rt-thread/components/finsh/finsh.h:60,
                 from /workspace/github/rtos/rt-thread/include/components.h:33,
                 from applications/application.c:23:
/usr/lib/gcc-cross/arm-linux-gnueabi/4.7/../../../../arm-linux-gnueabi/include/bits/pthreadtypes.h:165:22: note: previous declaration of 'pthread_spinlock_t' was here
In file included from /workspace/github/rtos/rt-thread/include/components.h:72:0,
                 from applications/application.c:23:
/workspace/github/rtos/rt-thread/components/pthreads/pthread.h:140:32: error: conflicting types for 'pthread_barrier_t'
In file included from /usr/lib/gcc-cross/arm-linux-gnueabi/4.7/../../../../arm-linux-gnueabi/include/sys/types.h:270:0,
                 from /workspace/github/rtos/rt-thread/components/finsh/finsh.h:60,
                 from /workspace/github/rtos/rt-thread/include/components.h:33,
                 from applications/application.c:23:
/usr/lib/gcc-cross/arm-linux-gnueabi/4.7/../../../../arm-linux-gnueabi/include/bits/pthreadtypes.h:174:3: note: previous declaration of 'pthread_barrier_t' was here
scons: *** [build/bsp/applications/application.o] Error 1
scons: building terminated because of errors.
```
很多这样的错，看着都是pthread 的问题；
我用的编译器是ubuntu 软件中心的`arm-linux-gnueabi-gcc`，我只修改了rtconfig.py里的`PREFIX =`其它都没动；CROSS_TOOL 和EXEC_PATH 都设置好了。

BernardXiong, 2016-01-16:
应该是pthread冲突了，可以把RT-Thread里的pthread关闭了。

oska874, 2016-01-16:
还是有问题，在`rtconfig.h` 注释掉`RT_USING_PTHREADS`之后编译提示找不到`reent.h`,然后又把`RT_USING_LIBC` 注释掉，编译有提示：
```
/usr/lib/gcc-cross/arm-linux-gnueabihf/4.8/../../../../arm-linux-gnueabihf/bin/ld: error: required section '.rel.plt' not found in the linker script
/usr/lib/gcc-cross/arm-linux-gnueabihf/4.8/../../../../arm-linux-gnueabihf/bin/ld: final link failed: Invalid operation
collect2: error: ld returned 1 exit status
scons: *** [rtthread-zynq7000.elf] Error 1
scons: building terminated because of errors.
```
不知道我哪一步有问题？

BernardXiong, 2016-01-16:
这个就要调整链接脚本了，这个和你用的编译器密切相关

oska874, 2016-01-16:
我用的`arm-linux-gnueabi-gcc 4.7` 和`arm-linux-gnueabihf-gcc 4.8`，以前编译其它rtos 并没有用到`.rep.plt` 段。
我又看了看和zynq 有关的代码，并没有发现哪里用到了`rel.plt`，虽然`src/module` 里面
```
/* locate .rel.plt and .rel.dyn section */
symtab = (Elf32_Sym *)((rt_uint8_t *)module_ptr +
                       shdr[shdr[index].sh_link].sh_offset);

```
里面提到了，但是zynq 的rtconfig 并没有打开MODULE，很奇怪。到这里为什么会出问题？

如果是编译器的问题，你们用的是什么编译器来编译zynq7000 的？

BernardXiong, 2016-01-18:
我这边用得比较多的是code sourery g++ lite吧

grissiom, 2016-01-19:
Zynq 的工具链推荐用 Xilinx 的：

http://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/vivado-design-tools.html

里面有 SDK，可以进行软件开发，而且貌似这个 SDK 是免费的～

BernardXiong, 2016-01-19:
嗯， 这个也可以试试。这个按照道理是可以的，因为有用整套vivado来编译RTT的，但是scons命令行编译可能会有出入。

closed 2016-04-02 by BernardXiong

liubenyuan, 2016-04-15:
@BernardXiong @grissiom 

事实上，典型的RTOS在Xilinx Zynq上的应用例程是

 - 用standalone bsp编译出fsbl.elf
 - 将Xilinx Zynq中FPGA的bit文件放置好
 - 编译出app.elf

最后，将三者链接起来制作成BOOT.bin（放在SD卡中启动）或者BOOT.mcs（烧写进Flash中启动）。一般用SDK比较方便，并且针对Zynq系列，Vivado和SDK都是免费的。

我很想帮助RTThread更新下Xilinx Zynq的教程，但是演示demo需要用到lwip，不知道RTThread的lwip支持怎么样？尤其是与Xilinx DMA对接的代码有没有处理好呢？

另外我看Zynq的bsp有两年没更新了，我准备fork一下，看能不能搞定

oska874, 2016-04-15:
@liubenyuan lwip 并不依赖硬件，纯软件+网卡驱动即可。

BernardXiong, 2016-04-15:
@liubenyuan zynq里是两个Cortex-A9，你上面说的两个elf，分别对应两个A9？

RT-Thread上你只要把以太网驱动对接好，就可以使用RT-Thread带的lwIP了。用rtconfig.h来做管理、裁剪的

BernardXiong, 2016-04-15:
@liubenyuan 你有其他联系方式吗（QQ、微信、邮箱）？我可以提供一份Xilinx SDK环境下的 RT-Thread 移植。

liubenyuan, 2016-04-15:
@BernardXiong 有的。微信liubenyuan1223，邮箱lbyoopp@163.com

liubenyuan, 2016-04-15:
@oska874 恩，我之前做过一个freertos_zynq_bsp（很原始，就是照着xilinx xapp1026改的）

https://github.com/liubenyuan/freertos_zynq_bsp

很想尝试一下rt-thread。一起努力：）

oska874, 2016-04-17:
@liubenyuan  xilinx 提供了freertos 的例程，的确很方便。

oska874, 2016-08-25:
@BernardXiong 现在用xilinx的提供的gcc 配合 scons 编译还是不通过，提示找不到 reent.h。

BernardXiong, 2016-08-26:
@oska874 
实际上你可以查看rt-thread的自动编译脚本，zynq的默认编译检查是放在里面的。从这个方式，你可以反推如何去搭建rt-thread的开发环境：
  - RTT_BSP='zynq7000' RTT_TOOL_CHAIN='sourcery-arm'
https://github.com/RT-Thread/rt-thread/blob/master/.travis.yml

oska874, 2016-08-26:
@BernardXiong 算了，放弃了。

liubenyuan, 2016-09-19:
@BernardXiong @oska874 

进展如何，最近腾出两个月，想把现在freertos替换为zynq