RT-Thread/rt-thread #611: 建议在设备接口底层增加init/deinit功能的接口，在设备关闭的时候省电、省电、省电（强烈建议）

heyuanjie87, 2016-02-25:


BernardXiong, 2016-02-25:
给出具体说明吧

另外，底层设备接口本身就有open/close操作，这个里面也可以关闭设备时钟，电源。

heyuanjie87, 2016-02-26:
我用串口操作一个模块，在不用的时候直接拉闸顺便close device，目前的处理是关闭中断，如果在这里关闭时钟感觉不太规范。

pangweishen, 2016-03-01:
有个ioctl的接口

heyuanjie87, 2016-03-02:
建议底层增加init(rt_device_t dev, int state)实现打开关闭的操作

grissiom, 2016-03-02:
`init(rt_device_t dev, int state)` 看起来是初始化的，不是开关闭的……

heyuanjie87, 2016-03-03:
这个是借鉴stm32标准外设库的init/deinit，为了省个接口用参数表示操作类型。至于接口的名称和使用方式可以再商量嘛，我只希望能增加 开启-关闭 外设的操作

BernardXiong, 2016-03-03:
开启、关闭就open/close啊

heyuanjie87, 2016-03-04:
close后并没有关闭外设控制器，我希望能把外设控制器也关了

BernardXiong, 2016-03-04:
你在底层驱动的close接口中处理

heyuanjie87, 2016-03-07:
struct rt_uart_ops
{
    rt_err_t (*configure)(struct rt_serial_device *serial, struct serial_configure *cfg);
    rt_err_t (*control)(struct rt_serial_device *serial, int cmd, void *arg);
    int (*putc)(struct rt_serial_device *serial, char c);
    int (*getc)(struct rt_serial_device *serial);
    rt_size_t (*dma_transmit)(struct rt_serial_device *serial, const rt_uint8_t *buf, rt_size_t size, int direction);
	void (*init)(struct rt_serial_device *serial, int state);
};
这个结构体改成这样可好。

对于grissiom看法:“init(rt_device_t dev, int state) 看起来是初始化的，不是开关闭的……”。
我的辩驳理由：一个是初始化为工作状态，一个是初始化为复位状态。

grissiom, 2016-03-07:
On Mon, Mar 7, 2016 at 9:26 AM, heyuanjie87 <notifications@github.com>
wrote:
>
> 对于grissiom看法:“init(rt_device_t dev, int state) 看起来是初始化的，不是开关闭的……”。
> 我的辩驳理由：一个是初始化为工作状态，一个是初始化为复位状态。
>
嗯，那还有更多状态呢，比如初始化成STREAM状态，初始化成中断状态，初始化成DMA状态，是不是每一个都要分配一个 state？

-- 
Cheers,
Grissiom

heyuanjie87, 2016-03-08:
这。。。。。。。
本来是借鉴下init/deinit并秉承rtt的节俭精神将它们人剑合一的。没想到还可以有这么多解释！

grissiom, 2016-03-08:
2016-03-08 9:23 GMT+08:00 heyuanjie87 <notifications@github.com>:

> 这。。。。。。。
> 本来是借鉴下init/deinit并秉承rtt的节俭精神将它们人剑合一的。没想到还可以有这么多解释！
>
>
人剑合一不是看起来人剑合一，而是一眼就能从头看到尾的合一。

Keep It Simple, Stupid.

-- 
Cheers,
Grissiom

heyuanjie87, 2016-03-09:
熊老大出来主持公道到底咋整,ioctl不赞成，我已经两次因为需要去添加关闭接口了。

BernardXiong, 2016-03-09:
呃，要不我们找个时间在网上音频pk下吧，我还没弄明白啥意思。

heyuanjie87, 2016-03-10:
在以前的处理中关闭设备仅禁止了中断，为了省电我希望能把外设控制器也禁用掉，目前没提供这样的底层接口。

wuji2015, 2016-04-06:
你在close里添加关闭外设函数的不可以（就是由close主动调用）？单独把那个列出来会不会破坏原来的统一标准。

bright-pan, 2016-07-14:
有点意思，对于开发者来说，是给鱼还是给渔是个问题，好久没有关注这里了

mysterywolf, 2021-08-06:
https://club.rt-thread.org/ask/question/432345.html 5年之后又有人提出这个问题了

grissiom, 2021-08-11:
我的理解应该是 driver 在 close 的时候需要调用 HAL_xxx_DeInit

mysterywolf, 2022-04-30:
https://www.rt-thread.org/document/site/#/rt-thread-version/rt-thread-standard/programming-manual/device/device?id=%e5%88%9b%e5%bb%ba%e5%92%8c%e6%b3%a8%e5%86%8c-io-%e8%ae%be%e5%a4%87

确实的，init方法会永久将RT_DEVICE_FLAG_ACTIVATED位置1，即便close掉了，也依然标明这个外设被激活了，并没有对应deinit方法将RT_DEVICE_FLAG_ACTIVATED重新置0

closed 2022-12-08 by heyuanjie87