RT-Thread/rt-thread #627: GCC 链接失败

tcz717, 2016-04-29:
当使用`GCC`编译`stm32f407`工程时 如果定义了**`RT_USING_LIBC`**以及**`RT_USING_DFS_DEVFS`**时在最后的连接过程会出错。

> LINK rtthread-stm32f4xx.axf
> /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: error: rtthread-stm32f4xx.axf uses VFP register arguments, /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-atoi.o) does not
> /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-atoi.o)
> /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: error: rtthread-stm32f4xx.axf uses VFP register arguments, /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-impure.o) does not
> /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-impure.o)
> /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: error: rtthread-stm32f4xx.axf uses VFP register arguments, /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-memcmp.o) does not
> /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-memcmp.o)
> /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: error: rtthread-stm32f4xx.axf uses VFP register arguments, /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-memcpy-stub.o) does not
> /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-memcpy-stub.o)
> /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld: error: rtthread-stm32f4xx.axf uses VFP register arguments, /opt/arm-2014.05/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-memset.o) does not
>......

BernardXiong, 2016-05-02:
应该是编译参数不匹配，建议查查编译参数。

tcz717, 2016-05-02:
编译参数是直接复制的bsp 唯一的变化就是加了c99

CornerOfSkyline, 2017-08-15:
@tcz717 这个问题解决了吗？  我发现应该是编译工具的问题，我查看了下..\arm-2014.05\arm-none-eabi\lib\thumb2下的libc.a文件大小为0KB，这个文件应该有问题

tcz717, 2017-08-15:
@CornerOfSkyline 解决了，过的有点久 我就记得好像是换了个什么东西

mysterywolf, 2021-02-06:
2021-2-7 最新的STM32F407工程不会出现该问题了

closed 2021-02-06 by mysterywolf