RT-Thread/rt-thread #629: RT_ASSERT

heyuanjie87, 2016-05-10:
在这个语句后最好添加判断语句，这样方便出问题后返回到调用的地方。

BernardXiong, 2016-05-10:
这个是个坑，要处理好不容易。

例如，
uint8_t *ptr1 = malloc(11);
RT_ASSERT_RET(ptr1 != NULL, -1);

uint8_t *ptr2 = malloc(22);
RT_ASSERT_RET(ptr2 != NULL, -1);

然后这个时候ptr1怎么办？

heyuanjie87, 2016-05-11:
不能这样偷懒，上次看到mfc里面的处理方式是这样的
uint8_t *ptr1 = malloc(11);
RT_ASSERT(ptr1 != NULL);
if (ptr1 == NULL)
{
    return -1;
}
uint8_t *ptr2 = malloc(22);
RT_ASSERT(ptr2 != NULL);
if (ptr2 == NULL)
{
    free(ptr1);
    return -1;
}

目前要改的地方挺多的，可以先把内核部分修正下

heyuanjie87, 2016-05-11:
RT_ASSERT_RET就没必要有了，最终目的是要保证在禁用ASSERT的情况下程序不轻易崩溃。

BernardXiong, 2016-05-12:
这样也不见得好，会增加挺多代码的。如果是能够确定代码没问题，可以把RT_ASSERT定义成空，这样指令会精简。

或者说对于assert进行分等级

heyuanjie87, 2016-05-13:
代码量是会增加一些，但是这可以提高稳定性 方便调试。最近就遇到这样的麻烦了ASSERT后不方便找出来在哪出的问题。

armink, 2016-05-14:
> 在这个语句后最好添加判断语句，这样方便出问题后返回到调用的地方。

@heyuanjie87 你说的这点我不太赞同，本身断言的目的就是在开发阶段检查软件逻辑性错误，逻辑性的错误说明代码编写就有问题，直接输出断言信息提示调用者，而不是让软件继续运行。

正好我最近编写了一份公司内部的断言使用规范文档，大家可以一起交流下，[点击这里阅读](https://www.zybuluo.com/armink/note/376419) 。

heyuanjie87, 2016-05-14:
@armink 事情是这样的，且听我娓娓道来：

void func()
{
    ASSERT(0);
}

我的代码中很多地方都调用了func。出问题后我只知道是func除了问题，要想知道在哪里调用出的问题就麻烦了。

armink, 2016-05-14:
那可以在出错时，把当前栈信息打印出来，分析查找函数调用栈信息。如果 Linux 平台，还可以通过 glibc 自带的 execinfo.h 头文件中现成的 backtrace 方法，输出函数调用栈的详细信息，包括函数名。

所以你的问题跟断言的关系不大，只是想要在异常处理时，定位错误原因。

closed 2021-02-28 by mysterywolf