RT-Thread/rt-thread #631: LWIP

heyuanjie87, 2016-05-13:
1.   
         if (!(ethif->flags & ETHIF_LINK_PHYUP))
            {
                netif_set_link_up(ethif->netif);
            }
这行代码是不是有bug？

2.
set_if这个接口应该开放出来，在读取配置文件设置静态ip的应用中会用到。

3.
netif_find返回值是NULL

BernardXiong, 2016-05-14:
写得清楚些

heyuanjie87, 2016-05-14:
1.
if (!(ethif->flags & ETHIF_LINK_PHYUP)) 
{
 netif_set_link_up(ethif->netif);
 }
本来想用netif->flags检测网线连接状态的，结果没插网线就发现网络连接标记值位了。在lwip中找出两处这样的代码。
是否应该写成：
if (ethif->flags & ETHIF_LINK_PHYUP) 
{
 netif_set_link_up(ethif->netif);
 }

2.
set_if这个接口是在使用finsh的时候导出的，希望取消这个限制。

3.
netif_find改成这样执行正确：
struct netif *
netif_find(char *name)
{
  struct netif *netif;
  u8_t num;

  if (name == NULL) {
    return NULL;
  }

  num = name[2] - '0';

  for(netif = netif_list; netif != NULL; netif = netif->next) {
    if (strncmp(name, netif->name, sizeof(netif->name)) == 0) {
      LWIP_DEBUGF(NETIF_DEBUG, ("netif_find: found %c%c\n", name[0], name[1]));
      return netif;
    }
  }
  LWIP_DEBUGF(NETIF_DEBUG, ("netif_find: didn't find %c%c\n", name[0], name[1]));
  return NULL;
}

原版
struct netif *
netif_find(char *name)
{
  struct netif *netif;
  u8_t num;

  if (name == NULL) {
    return NULL;
  }

  num = name[2] - '0';

  for(netif = netif_list; netif != NULL; netif = netif->next) {
    if (num == netif->num &&
       name[0] == netif->name[0] &&
       name[1] == netif->name[1]) {
      LWIP_DEBUGF(NETIF_DEBUG, ("netif_find: found %c%c\n", name[0], name[1]));
      return netif;
    }
  }
  LWIP_DEBUGF(NETIF_DEBUG, ("netif_find: didn't find %c%c\n", name[0], name[1]));
  return NULL;
}

BernardXiong, 2016-05-17:
netif_find改不动，这个lwIP原版就是这样。

set_if这个后续考虑整理下API接口了，也许ifconfig更通用些，但还没想好更好的方式。

UP那个，默认是会考虑让网卡UP起来。具体可能需要仔细看看

closed 2022-03-28 by Guozhanxin