RT-Thread/rt-thread #643: 升级STM32底层库函数？

Linjieqiang, 2016-07-05:
感谢你们的付出，我用的是STM405进行项目的编写，我看目前底层库函数是1.0.0，而当前最新的1.7.0，我尝试更换，可惜失败……能否定期更新支持芯片的库函数？

BernardXiong, 2016-07-05:
更新支持依赖的是大家对RT-Thread的支持，你可以尝试进行更新，如果遇到问题也可以发上来，大家会提供相应的一些建议。更新成功后，也欢迎把更新的版本提交到github上来

Linjieqiang, 2016-07-06:
好吧 我是初学者 还在学习实时操作系统 我看看 主要是替换后冲突太多 或者报错太多 我得仔细看看如何构成

Linjieqiang, 2016-07-06:
[image]
[image]
我刚试了试，keil能编译成功，就是scons编译不过去。。。。

BernardXiong, 2016-07-06:
f40x没有fmc吧，还是fsmc。

Linjieqiang, 2016-07-06:
嗯嗯，是的，f40x没有fmc，然后我对scons不太熟悉，不知道宏定义在哪……后来我重新用keil做了个最小系统，搞好了，谢谢。删掉一些无用的资源文件，然后清爽了，然后打算用它来写飞控，感谢你们团队的系统，学习中，资料很不错~！
[image]

closed 2016-07-06 by Linjieqiang