RT-Thread/rt-thread #673: Manual in english

eskomj, 2016-10-12:
Hello,

There is manual in english in github, but could somebody make english version available? It could help understanding of rt-thread a lot.

Esko Jaakkola
Finland

BernardXiong, 2016-10-12:
Normally, you can use Google to translate RT-Thread programming guide. Moreover, the API manual of RT-Thread is in English.

wzyy2, 2016-10-13:
I think RT-Thread should have a wiki for manual or guide.
It will help a lot.

BernardXiong, 2016-10-13:
Yes, and the wiki can be placed on github.

eskomj, 2016-10-15:
I found typo in my original message. Of course I meant that there is chinese version available. It really could help, if someone who manage chinese and english could do translation. Wiki would be best solution.

mysterywolf, 2021-02-05:
https://www.rt-thread.io/document/site/

closed 2021-02-05 by mysterywolf