RT-Thread/rt-thread #679: Created lwIP repo with NAT ported to latest git version

ajaybhargav, 2016-11-19:
Cleaned up and ported to latest git version. I have also tested it on my test board. NAT works great. Thanks for making this possible.
https://github.com/ajaybhargav/lwip_nat

Regards,
Ajay Bhargav

BernardXiong, 2016-11-20:
Yes, I found it. If any updates, please feed back to RT-Thread thanks!

BTW, I just ported the lwIP 2.0.0 version to RT-Thread.

ajaybhargav, 2016-11-21:
Yes I will, I have did some cleanup in the NAT code so that it complies to lwIP guidelines. Also thanks for responding to lwIP email thread. Once the license issue gets resolved, I will submit matches to lwIP. NAT will then be officially supported in lwIP. I will also keep an eye on RT-Thread for any updates in NAT source. You can try to integrate my lwIP source in RT-Thread as submodule maybe. I will keep the repo updated to lwIP head revision.

ajaybhargav, 2016-12-28:
Is there any update on licensing of code? If Author permits dual license then it might solve the issue. We can mainline NAT code.

BernardXiong, 2016-12-28:
@ajaybhargav 
There is no license issue for NAT code. I have asked @Hichard, there is no license issue.

closed 2017-06-15 by BernardXiong