RT-Thread/rt-thread #696: rt_vsnprintf does not support %f and %lf

hesst, 2017-02-09:
float and double are not supported by rt_vsnprintf

BernardXiong, 2017-02-10:
Yes. In order to reduce footprint, there is no float/double implementation in rt_sprintf/vsnprintf and rt_kprintf. You can use the libc's, such as vsnprintf to a buffer, then print out to the console if you need the log.

hesst, 2017-02-10:
I always have defined RT_USING_LIBC.
With gnu gcc rt-thread defines then RT_USING_NEWLIB and then snprintf is aliased to rt_snprintf
```
int snprintf(char *buf, rt_size_t size, const char *fmt, ...) __attribute__((weak, alias("rt_snprintf")));
int vsprintf(char *buf, const char *format, va_list arg_ptr) __attribute__((weak, alias("rt_vsprintf")));
```

BernardXiong, 2017-02-11:
Oh, It's a bug in RT-Thread. You can submit a patch, thanks.

BernardXiong, 2017-04-08:
@hesst 

I have checked the code, the alias should be not used:
#if !defined (RT_USING_NEWLIB) && defined (RT_USING_MINILIBC) && defined (__GNUC__)
....
#endif

hesst, 2017-04-10:
I see, but then I dont understand why a call to snprintf results in a call to rt_snprintf.
I have a module with dependency too snprintf:
```
Relocation section '.rel.plt' at offset 0x93c contains 39 entries:
 Offset     Info    Type            Sym.Value  Sym. Name
00002164  00000716 R_ARM_JUMP_SLOT   00000000   snprintf
```
In components/libc/newlib/libc_sym.c snprintf is defined as exportet symbol:
`RTM_EXPORT(snprintf);`
I can find this symbol in the elf file
```
  6711: 00000000     0 FILE    LOCAL  DEFAULT  ABS snprintf.c
  6876: 00000000     0 FILE    LOCAL  DEFAULT  ABS vsnprintf.c
  7417: 301544b8    13 OBJECT  GLOBAL DEFAULT    3 __rtmsym_rt_vsnprintf_nam
  **7635: 300fa4ec   228 FUNC    GLOBAL DEFAULT    1 snprintf**
  8072: 3000d7f4    80 FUNC    GLOBAL DEFAULT    1 rt_snprintf
  8488: 3000cf2c  2248 FUNC    GLOBAL DEFAULT    1 rt_vsnprintf
  8512: 3010eab0     8 OBJECT  GLOBAL DEFAULT    1 __rtmsym_rt_snprintf
  8822: 301544c8    12 OBJECT  GLOBAL DEFAULT    3 __rtmsym_rt_snprintf_name
  9157: 300fa418   212 FUNC    GLOBAL DEFAULT    1 _snprintf_r
  9328: 3010eaa8     8 OBJECT  GLOBAL DEFAULT    1 __rtmsym_rt_vsnprintf
 **10054: 3010fa80     8 OBJECT  GLOBAL DEFAULT    1 __rtmsym_snprintf**
 10283: 30102e48    56 FUNC    GLOBAL DEFAULT    1 vsnprintf
 **10452: 30157960     9 OBJECT  GLOBAL DEFAULT    3 __rtmsym_snprintf_name**
 10850: 30102d90   184 FUNC    GLOBAL DEFAULT    1 _vsnprintf_r
```
at address 0x300fa4ec is snprintf from gnu libc
```
 .text          0x300fa418      0x1b8 d:/sysgcc/arm-eabi/bin/../lib/gcc/arm-eabi/6.2.0/../../../../arm-eabi/lib\libc.a(lib_a-snprintf.o)
                0x300fa418                _snprintf_r                                                                                              
                0x300fa4ec                snprintf  
```

BernardXiong, 2017-04-10:
What's your environment? STM32 + scons under Linux? If you use scons, the RT_USING_NEWLIB should be defined. 

Or you can use `scons --verbose` to build rt-thread.

hesst, 2017-04-10:
mini2440 +scons under win7. RT_USING_NEWLIB is defined

BernardXiong, 2017-04-10:
-_- which rt-thread version? What's the logs for `scons --verbose`? It's there the `-DRT_USING_NEWLIB`?

hesst, 2017-04-10:
> -_- which rt-thread version? 
A copy of the master-branch from december 24, 2016.
I use the toolchain from this site: http://gnutoolchains.com/download/
the first
`2.27 	6.2.0 	Newlib 2.4.0.20160923 	7.12 	arm-eabi-gcc6.2.0-r3.exe
`
I have added the -nostartfiles to the LFLAGS in rtconfig.py, because -u,_start doesn't worked for me with this toolchain.

> What's the logs for scons --verbose? 
for the first file of rt-thread firmware:
`arm-eabi-gcc -o build\applications\application.o -c -mcpu=arm920t -O0 -gstabs+ -DRT_THREAD -DRT_USING_NEWLIB -I. -Iapplications -Idrivers -ID:\rt-thread\include -ID:\rt-thread\libcpu\arm\s3c24x0 -ID:\rt-thread\libcpu\arm\common -ID:\rt-thread\components\dfs\include -ID:\rt-thread\components\dfs\filesystems\devfs -ID:\rt-thread\components\dfs\filesystems\elmfat -ID:\rt-thread\components\drivers\include -ID:\rt-thread\components\drivers\include -ID:\rt-thread\components\external\ftk\ftk\src -ID:\rt-thread\components\external\ftk\ftk\src\canvas\default -ID:\rt-thread\components\external\ftk\ftk\apps\common -ID:\rt-thread\components\external\ftk\ftk\src\os\rt-thread -ID:\rt-thread\components\external\ftk\ftk\src\linebreak -ID:\rt-thread\components\external\ftk\ftk\src\demos -ID:\rt-thread\components\external\ftk\ftk\src\backend\rt-thread -ID:\rt-thread\components\external\jpeg -ID:\rt-thread\components\external\libpng -ID:\rt-thread\components\external\libz -ID:\rt-thread\components\finsh -ID:\rt-thread\components\libc\newlib -ID:\rt-thread\components\libc\pthreads -ID:\rt-thread\components\libc\libdl -ID:\rt-thread\components\net\lwip-1.4.1\src -ID:\rt-thread\components\net\lwip-1.4.1\src\include -ID:\rt-thread\components\net\lwip-1.4.1\src\include\ipv4 -ID:\rt-thread\components\net\lwip-1.4.1\src\arch\include -ID:\rt-thread\components\net\lwip-1.4.1\src\include\netif applications\application.c`

for the first file of module:
`arm-eabi-gcc -o calculator\build\mini2440\main.o -c -mcpu=arm920t -O0 -fPIC -DRT_USING_NEWLIB -DFTK_AS_PLUGIN -DRT_THREAD -O0 -gstabs+ -ID:\rt-thread\bsp\mini2440 -ID:\rt-thread\bsp\mini2440\applications -ID:\rt-thread\bsp\mini2440\drivers -ID:\rt-thread\components\dfs\filesystems\devfs -ID:\rt-thread\components\dfs\filesystems\elmfat -ID:\rt-thread\components\dfs\include -ID:\rt-thread\components\drivers\include -ID:\rt-thread\components\external\jpeg -ID:\rt-thread\components\external\ftk\ftk\src -ID:\rt-thread\components\external\ftk\ftk\src\canvas\default -Icommon -ID:\rt-thread\components\external\ftk\ftk\src\os\rt-thread -ID:\rt-thread\components\external\ftk\ftk\src\linebreak -ID:\rt-thread\components\external\ftk\ftk\src\demos -ID:\rt-thread\components\external\ftk\ftk\src\backend\rt-thread -ID:\rt-thread\components\finsh -ID:\rt-thread\components\libc\libdl -ID:\rt-thread\components\libc\newlib -ID:\rt-thread\components\libc\pthreads -ID:\rt-thread\components\net\lwip-1.4.1\src -ID:\rt-thread\components\net\lwip-1.4.1\src\arch\include -ID:\rt-thread\components\net\lwip-1.4.1\src\include -ID:\rt-thread\components\net\lwip-1.4.1\src\include\ipv4 -ID:\rt-thread\components\net\lwip-1.4.1\src\include\netif -ID:\rt-thread\include -ID:\rt-thread\libcpu\arm\common -ID:\rt-thread\libcpu\arm\s3c24x0 calculator\main.c`

do you need the whole log?

>> Is it there the -DRT_USING_NEWLIB?

> RT_USING_NEWLIB is defined

BernardXiong, 2017-04-10:
Maybe you can add somethings inside:
#if !defined (RT_USING_NEWLIB) && defined (RT_USING_MINILIBC) && defined (GNUC)
....
#endif

to make sure it's compiled. And if you use application module, please add RT_USING_NEWLIB definition.

hesst, 2017-04-10:
I can try, but this can not go into the compilation, because RT_USING_NEWLIB is defined.
RT_USING_NEWLIB is also defined in the module.

I dont think anymore the snprintf alias is the problem. 
The reason why I thought this, was a search over all sources.
The NOT in #if !defined (RT_USING_NEWLIB) && defined (RT_USING_MINILIBC) && defined (GNUC) I've overlooked.
The reason must be somewhere else. The search for snprintf when loading the module also looks not as rt_snprintf could be found.

BernardXiong, 2017-04-10:
First of all, please make sure somethings,
1. in the Kernel, if you invoke snprintf, whether the rt_snprintf has been called?

2. in the Application Module, if you invoke snprintf, whether the rt_snprintf has been called?

hesst, 2017-04-11:
Ok, in 2 weeks at weekend I'm at home and can try it.

closed 2021-02-24 by mysterywolf