RT-Thread/rt-thread #712: 删除已创建但未启动的线程时，可能会导致系统瘫痪

armink, 2017-04-08:
## 1、原因分析

在线程对象中有一个 `number_mask` 属性，该属性在 **线程启动** 时会进行初始化。如果线程 **未启动** 就立刻删除线程时，在 [rt_thread_delete->rt_schedule_remove_thread (scheduler.c#L348)](https://github.com/RT-Thread/rt-thread/blob/master/src/scheduler.c#L348) 中会使用该属性，如下：

```C
void rt_schedule_remove_thread(struct rt_thread *thread)
{
   ......
    if (rt_list_isempty(&(rt_thread_priority_table[thread->current_priority])))
    {
#if RT_THREAD_PRIORITY_MAX > 32
        rt_thread_ready_table[thread->number] &= ~thread->high_mask;
        if (rt_thread_ready_table[thread->number] == 0)
        {
            rt_thread_ready_priority_group &= ~thread->number_mask;
        }
#else
        rt_thread_ready_priority_group &= ~thread->number_mask;
#endif
    }
```

此时由于 thread->number_mask 为随机值，那么 `rt_thread_ready_priority_group ` 就会被改坏，甚至有可能是 **0**。等到下次调度时，计算出来的 highest_ready_priority 就会出错，当 `rt_thread_ready_priority_group == 0` 时，计算出来的 highest_ready_priority 甚至为 0xFFFFFFFF ，这下取出 `to_thread` 就出现了错误，系统此时就无法工作了，相关代码如下：

```C
void rt_schedule(void)
{
        ......
        register rt_ubase_t highest_ready_priority;

#if RT_THREAD_PRIORITY_MAX <= 32
        highest_ready_priority = __rt_ffs(rt_thread_ready_priority_group) - 1;
#else
        ......
#endif

        /* get switch to thread */
        to_thread = rt_list_entry(rt_thread_priority_table[highest_ready_priority].next,
                                  struct rt_thread,
                                  tlist);
```

## 2、问题修复

个人的两个思路如下，欢迎大家一起交流，分享你的观点，确定一种修复方法。

### 2.1 思路1

线程属性 number_mask 放在线程 init 及 create 时，就立刻初始化。

### 2.2 思路2

函数 rt_schedule_remove_thread 中的变量 rt_thread_ready_priority_group 在计算时，不要使用 number_mask，而是直接通过线程优先级属性计算出来，如下：

```
rt_thread_ready_priority_group &= ~(1L << thread->current_priority);
```

## 3、问题延伸

### 3.1 系统关键属性修改时，缺少有效性检测

像 `rt_thread_ready_priority_group` 及 `highest_ready_priority` 这类关键性的变量在修改前，可以先进行有效性检测，如果不符合要求，断言提示出来，或者自动忽略、修复等，最大程度的提高系统的容错能力。

### 3.2 系统瘫痪时，定位问题的难度较大

此次问题的现象是产品偶尔会提示线程堆栈溢出，但实际上并不是这个原因，是一个很长的连锁反应，最终导致的这个现象。对于用户来说，定位问题的复杂度过高。所以，系统提示系统还可以再强壮一些，这样也便于更好的暴露出根本原因。

BernardXiong, 2017-04-08:
删除未启动的线程，可能还真是。因为都会假定它是一个已启动，已运行的线程。

可能可以针对线程状态做个检查。

armink, 2017-04-08:
rt_thread_detach 也存在这个问题。

那我就在删除时，对调度部分代码做个检查，改为下面的方式了：

```
rt_err_t rt_thread_delete(rt_thread_t thread)
{
    rt_base_t lock;

    /* thread check */
    RT_ASSERT(thread != RT_NULL);

    /* check stat */
    if(thread->stat != RT_THREAD_INIT)
    {
        /* remove from schedule */
        rt_schedule_remove_thread(thread);

        /* release thread timer */
        rt_timer_detach(&(thread->thread_timer));
    }

    /* change stat */
    thread->stat = RT_THREAD_CLOSE;
```

BernardXiong, 2017-04-08:
是的，对任务状态做检查是最简单的方式

grissiom, 2017-04-08:
线程初始化之后，各 mask 的值为啥不清零？

grissiom, 2017-04-10:
@armink 的 patch 有个问题，就是在判断线程状态的时候需要禁用中断…… 因为在调度器中本来就是禁用中断的，我倒是觉得修复掉 mask 的初始值是更好的做法。

armink, 2017-04-10:
@grissiom 大概明白你的意思了。刚发现，直接这样修复，遗漏了

```
rt_list_remove(&(thread->tlist));
```

这个工作以前是在 rt_schedule_remove_thread() 中完成的。所以，从解决 mask 的初始值问题入手，会更直接。

另外，感觉修复掉 mask 的初始化后还有个问题，在 `rt_schedule_remove_thread()` 中会将待删除线程的优先级从 rt_thread_ready_priority_group 中移除，如果此时该线程并没有启动过，但存在与其优先级相同的其他线程待调度，不应该一刀切。是不是也得加个判断，对未启动线程限制下？

BernardXiong, 2017-04-10:
嗯，可能会连带出系列的问题，我仔细考虑下吧

grissiom, 2017-04-12:
如果各个 mask 的初始值都是0，那么`rt_thread_ready_priority_group &= ~thread->number_mask`不会产生任何副作用。

armink, 2017-04-24:
@BernardXiong @grissiom `_rt_thread_init` 增加下面的 初始化 mask 代码后，真机测试目前也没有问题。

```C
    /* initialize mask */
    thread->number_mask = 0;

#if RT_THREAD_PRIORITY_MAX > 32
    /* initialize number and high mask */
    thread->number = 0;
    thread->high_mask = 0;
#endif
```

BernardXiong, 2017-06-15:
#8437c04 fix it.

closed 2017-06-15 by BernardXiong