RT-Thread/rt-thread #719: LPC4337编译报错

canghaiwuhen, 2017-04-16:
LPC4337的bsp在mdk5编译报错

BernardXiong, 2017-04-17:
请使用scons生成工程，然后再进行编译

canghaiwuhen, 2017-04-17:
使用scons生成的，只不过只能生成mdk4的，用mdk5打开后编译报错

BernardXiong, 2017-04-17:
scons --target=mdk5 -s
如果报错，请把报错的信息提供出来

canghaiwuhen, 2017-04-17:
[image]

BernardXiong, 2017-04-17:
自行使用MDK5创建一个空文件、空Group的工程，可以自行把芯片型号，调试选项等配置，并这个工程文件保存为template.uvprojx。

然后再使用scons来生成工程

canghaiwuhen, 2017-04-17:
按照这个做法生成了新的工程，但用scons编译报错，在mdk上编译M0没问题，编译M4报错，看说明文档意思是要将M0编译出来的bin文件转成M0_CODE.h文件？
[image]

BernardXiong, 2017-04-17:
是的，M4是和M0关联的。

canghaiwuhen, 2017-04-17:
非常感谢~~在mdk中已经可以编译烧写了，但在命令行下，M0用scons编译还是报错，但在M4下只是有警告

BernardXiong, 2017-04-17:
都用生成工程方式吧，因为一些mdk版本更改、链接脚本更改等的缘故，scons可能没更新上

canghaiwuhen, 2017-04-17:
好的~~

canghaiwuhen, 2017-04-17:
另外 LPC4337的工程模板的PLL时钟用的180M，这个芯片应该设置成204M才是，等测试没问题我提交一个~~

closed 2017-04-17 by canghaiwuhen