RT-Thread/rt-thread #800: 非“/”目录挂载点不存在，是否需要在系统API中创建

emlslxl, 2017-08-12:
现在的情况是，挂载第二个分区时，必须要事先创建挂载点目录，否则第二分区会因为挂载点不存在而挂载失败，挂载代码如下：
```
dfs_mount("flash0", "/", "elm", 0, 0)   //挂载第一分区
mkdir("/sd0",0);    //创建第二分区挂载点目录
dfs_mount("sd0", "/sd0", "elm", 0, 0)   //挂载第二分区
```
在dfs_mount（）函数内部，有如下一段代码：
`/* Check if the path exists or not, raw APIs call, fixme */
    if ((strcmp(fullpath, "/") != 0) && (strcmp(fullpath, "/dev") != 0))
    {
        struct dfs_fd fd;
        //mkdir(fullpath,0);    //create other partition mount point
        if (dfs_file_open(&fd, fullpath, DFS_O_RDONLY | DFS_O_DIRECTORY) < 0)
        {
            rt_free(fullpath);
            rt_set_errno(-DFS_STATUS_ENOTDIR);

            return -1;
        }
        dfs_file_close(&fd);
    }
`
该段代码会检查到非根目录挂载，是否可以在此处创建非根目录挂载点（注释掉的一行内容）

BernardXiong, 2017-08-12:
是的，挂载的文件系统必须挂载到存在的目录上，这个和linux下的mount行为是一致的。

emlslxl, 2017-08-14:
明白了

closed 2017-08-14 by emlslxl