RT-Thread/rt-thread #815: rt_components_init()初始化后 调用list_thread()命令会卡死

emlslxl, 2017-08-18:
使用组件初始化，在init任务中调用rt_components_init()，会导致卡死，代码如下：
```
int rt_application_init()
{
    rt_thread_t tid;

    tid = rt_thread_create("init",
                        rt_init_thread_entry, RT_NULL,
                        2048, RT_THREAD_PRIORITY_MAX / 3, 20);
    if (tid != RT_NULL) rt_thread_startup(tid);

    return 0;
}

void rt_init_thread_entry(void *parameter)
{
    rt_thread_t tid;

    rt_platform_init();

    rt_components_init();
}
```
仿真查看，发现无限循环在object_name_maxlen()函数内，因为链表查找不到尽头。

不使用rt_components_init()函数进行初始化，则一切正常。

armink, 2017-08-20:
单步定位/变量监控下 thread list 在哪里被改坏了。

emlslxl, 2017-08-20:
是我使用有问题，原本的工程中没有使用rt_components_init()，所以在rtthread_startup()中执行finsh_system_init()，后来想尝试一下用组件初始化任务，调用rt_components_init()的时候finsh_system_init()又被执行了一次，导致了对thread list的破坏。谢谢armink。

closed 2017-08-20 by BernardXiong