RT-Thread/rt-thread #838: gui在不同控件同时刷新时可能出现刷新操作的丢失

spavilion, 2017-09-05:
dc.c 行1893
win->drawing --;
if (win->drawing == 0 && rtgui_graphic_driver_is_vmode() == RT_FALSE)
{
......

必须要等到win->drawing==0时才会发送RTGUI_EVENT_UPDATE_END，在多控件出现绘制时，如响应不及时，则会丢失部分操作。我现在修改了void rtgui_dc_end_drawing(struct rtgui_dc *dc)函数，如下：

struct update_rects_node
{
	struct rtgui_widget *owner;
	struct update_rects_node *next;
};
struct update_rects_node *update_list = RT_NULL;
void rtgui_dc_end_drawing(struct rtgui_dc *dc)
{
    struct rtgui_widget *owner;
    struct rtgui_win *win;

    RT_ASSERT(dc != RT_NULL);

    /* get owner */
    if (dc->type == RTGUI_DC_CLIENT)
        owner = RTGUI_CONTAINER_OF(dc, struct rtgui_widget, dc_type);
    else if (dc->type == RTGUI_DC_HW)
        owner = ((struct rtgui_dc_hw *)dc)->owner;
    else return ; /* bad DC type */
    /* get window */
    win = owner->toplevel;

    /* decrease drawing counter */
    win->drawing --;
    if (win->drawing == 0 && rtgui_graphic_driver_is_vmode() == RT_FALSE)
    {
#ifdef RTGUI_USING_MOUSE_CURSOR
        rt_mutex_release(&cursor_mutex);
        /* show cursor */
        rtgui_mouse_show_cursor();
#endif
				while(update_list)
				{
					struct update_rects_node *p = update_list;
					if((rtgui_rect_contains_point(&owner->extent, p->owner->extent.x1, p->owner->extent.y1)) ||
						(rtgui_rect_contains_point(&owner->extent, p->owner->extent.x1, p->owner->extent.y2)) ||
						(rtgui_rect_contains_point(&owner->extent, p->owner->extent.x2, p->owner->extent.y1)) ||
						(rtgui_rect_contains_point(&owner->extent, p->owner->extent.x2, p->owner->extent.y2)))
					{
						if (RTGUI_IS_WINTITLE(p->owner->toplevel))
						{
								/* update screen */
								rtgui_graphic_driver_screen_update(rtgui_graphic_driver_get_default(),
																									 &(p->owner->extent));
						}
						else
						{
								/* send to server for window update */
								struct rtgui_event_update_end eupdate;
								RTGUI_EVENT_UPDATE_END_INIT(&(eupdate));
								eupdate.rect = p->owner->extent;
								rtgui_server_post_event((struct rtgui_event *)&eupdate, sizeof(eupdate));
						}
					}
					
					update_list = update_list->next;
					rt_free(p);
				}
        if (RTGUI_IS_WINTITLE(win))
        {
            /* update screen */
            rtgui_graphic_driver_screen_update(rtgui_graphic_driver_get_default(),
                                               &(owner->extent));
        }
        else
        {
            /* send to server for window update */
            struct rtgui_event_update_end eupdate;
            RTGUI_EVENT_UPDATE_END_INIT(&(eupdate));
            eupdate.rect = owner->extent;
            rtgui_server_post_event((struct rtgui_event *)&eupdate, sizeof(eupdate));
        }
    }
		else
		{
			struct update_rects_node *node = rt_malloc(sizeof(struct update_rects_node));
			node->owner = owner;
			node->next = RT_NULL;
			if(update_list == RT_NULL)
			{
				update_list = node;
			}
			else
			{
				struct update_rects_node *p = update_list->next, *p1 = update_list; 
				while(p != RT_NULL)
				{
					p1 = p;
					p = p->next;
				}
				p1->next = node;
			}
		}

    dc->engine->fini(dc);
    rtgui_screen_unlock();
}

当出现多个绘制时，先记录下绘制的对象，等到win->drawing==0成立时，判断记录下来的控件是否在要更新的范围内，如果不在，则单独更新。
请问这样写是否会造成其他问题？

BernardXiong, 2017-09-05:
原来的问题，有测试用例不？

spavilion, 2017-09-06:
我使用的平台需要特定的设备，大概就是一个notebook上有两个tab，每个tab上有数个控件，用按钮控制tab切换，另有一个label不停的更新时间。当tab切换时会出现丢失RTGUI_EVENT_UPDATE_END的情况。

BernardXiong, 2017-09-06:
嗯，如果有什么方式能够直接沟通下就比较好了。QQ  or 微信？

spavilion, 2017-09-06:
好的，这是我的QQ号：123987355

rh-chen, 2017-09-26:
wow, it's good ...

closed 2021-02-06 by mysterywolf