RT-Thread/rt-thread #851: RTT目前不支持ARM Compiler 6.6?

JcZou, 2017-09-21:
使用MDK5.23自带的ARM Compiler 6.6，选择gnu99
编译RTT工程产生错误，如下
.\build\rtthread-stm32f4xx.axf: Error: L6218E: Undefined symbol __bss_end (referred from startup.o).
.\build\rtthread-stm32f4xx.axf: Error: L6218E: Undefined symbol __fsymtab_end (referred from shell.o).
.\build\rtthread-stm32f4xx.axf: Error: L6218E: Undefined symbol __fsymtab_start (referred from shell.o).
.\build\rtthread-stm32f4xx.axf: Error: L6218E: Undefined symbol __vsymtab_end (referred from shell.o).
.\build\rtthread-stm32f4xx.axf: Error: L6218E: Undefined symbol __vsymtab_start (referred from shell.o).
请问最新的RTT还不支持ARM Compiler 6.6编译器么

BernardXiong, 2017-09-23:
是自己建的工程？另外，为什么你的armcc 6.6会以GNU GCC方式进行编译，是否某些地方配置不正确？

JcZou, 2017-09-23:
现在已经解决了，是用的RTT的最新的stm32f40x的工程。 环境是MDK 5.23，选择的编译器是ARM Compiler6.6，它是基于LLVM的，所以默认的编译器宏是__GNUC__而不是__CC_ARM

按如下方式修改问题已经解决
**StarryDrone/Application/startup.c：**
 #elif __ICCARM__
  #pragma section="HEAP"
  #define STM32_SRAM_BEGIN    (__segment_end("HEAP"))
 +#elif __GNUC__
 +  /* For MDK Compiler6.6(GNUC), we still use MDK as IDE */
 +	extern int Image$$RW_IRAM1$$ZI$$Limit;
 +	#define STM32_SRAM_BEGIN    (&Image$$RW_IRAM1$$ZI$$Limit)

**StarryDrone/RTOS/components/finsh/shell.c**
 #elif defined (__GNUC__) || defined(__TI_COMPILER_VERSION__)
      /* GNU GCC Compiler and TI CCS */
 -    extern const int __fsymtab_start;
 -    extern const int __fsymtab_end;
 -    extern const int __vsymtab_start;
 -    extern const int __vsymtab_end;
 -    finsh_system_function_init(&__fsymtab_start, &__fsymtab_end);
 -    finsh_system_var_init(&__vsymtab_start, &__vsymtab_end);
 +		/* For MDK Compiler6.6(GNUC), we still use MDK as IDE */
 +		extern const int FSymTab$$Base;
 +    extern const int FSymTab$$Limit;
 +    extern const int VSymTab$$Base;
 +    extern const int VSymTab$$Limit;
 +    finsh_system_function_init(&FSymTab$$Base, &FSymTab$$Limit);

TanekLiang, 2017-11-12:
建议按照cmsis库的方式写。这样就不会混淆gcc和armcc 6.x了

#if defined __CC_ARM
  #define __SIMD32_TYPE int32_t __packed
  #define CMSIS_UNUSED __attribute__((unused))

#elif defined(__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050)
  #define __SIMD32_TYPE int32_t
  #define CMSIS_UNUSED __attribute__((unused))

BernardXiong, 2017-12-09:
这个问题已经修复？如果是的话，可以把PR发出来。

BernardXiong, 2020-08-25:
kernel-sample-0.1.0，不知道这份出自哪里

closed 2021-06-13 by mysterywolf

jhchance, 2022-06-29:
> 
这个问题我也遇到了，添加include<signal.h>就解决了，但我也遇到了一楼的链接错误问题。

> 请问，rtthread_simulator_v0.1.0，怎么默认不支持ac6啊？？编译均出错 kernel-sample-0.1.0/signal_sample.c(37): error: use of undeclared identifier 'SIGUSR1' rt_signal_install(SIGUSR1, thread1_signal_handler); ^ kernel-sample-0.1.0/signal_sample.c(38): error: use of undeclared identifier 'SIGUSR1' rt_signal_unmask(SIGUSR1); ^ kernel-sample-0.1.0/signal_sample.c(66): error: use of undeclared identifier 'SIGUSR1' rt_thread_kill(tid1, SIGUSR1); ^

sunlightxhw, 2023-08-03:
作为新手，看到了官网整洁的文字排列。一上来按照官网新手入门教程进行操作，就遇到这个编译问题：signal.c 文件中的SIG_**不识别 。着实感觉不爽。

wuwenyuanx, 2023-08-18:
rtt真是太难用了，四处是坑