RT-Thread/rt-thread #852: NV32F100X 系列芯片需要在CPSIE指令后添加多一个NOP指令

Quintin-Z, 2017-09-23:
在移植NV32F100X 系列芯片的过程中发现首次启动后会多次lockup复位，单步跟踪后发现都是在CPSIE指令后，无法进入PENDSV中断，而且复位报lockup内核错误，跟原厂沟通后，回复：M0+架构需要在CPSIE指令后等待一个指令，以保住PRIMAKE被置位。我估计会不会是流水线的设计，导致这个PRIMASK还没置位而中断就响应了，然后就发生了内核错误，导致lockup。

Quintin-Z, 2017-09-23:
至于是NV32这一家芯片设计的问题还是M0+架构的通病有待测试

Guozhanxin, 2023-03-24:
这个最好是所有芯片都修一下，但不加NOP，而应该改成加 ISB

aozima, 2023-03-24:
后面的代码都已有同步指令，这个NOP可以不用加了

```asm
    ; trigger the PendSV exception (causes context switch)
    LDR     r0, =NVIC_INT_CTRL
    LDR     r1, =NVIC_PENDSVSET
    STR     r1, [r0] ; 问题的关键在于这里的写入有write buffer存在，并不能保证在当前周期内一定成功

    ; restore MSP
    LDR     r0, =SCB_VTOR
    LDR     r0, [r0]
    LDR     r0, [r0]
    MSR     msp, r0

    ; enable interrupts at processor level
    CPSIE   F
    CPSIE   I

    ; ensure PendSV exception taken place before subsequent operation
    DSB
    ISB

    ; never reach here!
    ENDP
```

closed 2023-03-27 by Guozhanxin