RT-Thread/rt-thread #853: 使用scons无法编译RTT

greedyhao, 2017-09-24:
版本信息：python 2.7，scons 3.0
编译报错：
```
D:\File\DeveloperFile\MicroControllers\rt-thread-master\rt-thread-master\bsp\stm32f10x> scons
scons: Reading SConscript files ...
  File "D:\File\DeveloperFile\MicroControllers\rt-thread-master\rt-thread-master\components\external\cairo\SConscript", line 9

    print '================ERROR============================'

                                                            ^

SyntaxError: invalid syntax
```
我只更改了stm32f10x下的rtconfig.py文件，在网上搜索不到解决方案，官方文档也没有看到这个问题的解决方法，希望能指点我一下。

BernardXiong, 2017-09-24:
你安装的是Python 3.x吧

joyhope, 2017-09-28:
python 2.7 也同样。 linux, 和 windows 都是类似错误。 感觉怎么 def GetDepend(depend) 的函数工作不正常。

CornerOfSkyline, 2017-09-28:
scons不能装3.0版本的，安装2.5.1版本的 @joyhope

joyhope, 2017-09-28:
换到2.5.1 正常了。 需要在readme里面写一下，这样大家就不会上当了。默认scon 给了3.0.0下载。谢谢@CornorOfSkyline

greedyhao, 2017-10-10:
@CornerOfSkyline 非常感谢，这解决了我的问题。

closed 2017-10-10 by greedyhao