RT-Thread/rt-thread #1020: RT_USING_POSIX 是否和fish 使用有冲突

joyhope, 2017-11-20:
我打开了RT_USING_POSIX，也打开了RT_USING_FINSH

在STM32f10x_HAL 的bsp里面，出现
`.\build\rtthread-stm32.axf: Error: L6218E: Undefined symbol finsh_set_device (referred from startup.o).`

看来代码，的确是因为POSIX, 把这个函数关闭了。逻辑上这2者应该没有明显冲突，应该是可以同时使用的吧。

joyhope, 2017-11-20:
我猜测，做下面一个简单的实现
`void finsh_set_device(const char *device_name)
{
    rt_device_t dev = RT_NULL;

    RT_ASSERT(shell != RT_NULL);
    dev = rt_device_find(device_name);
    if (dev == RT_NULL)
    {
        rt_kprintf("finsh: can not find device: %s\n", device_name);
        return;
    }

    /* open this device and set the new device in finsh shell */
    if (rt_device_open(dev, RT_DEVICE_OFLAG_RDWR | RT_DEVICE_FLAG_INT_RX | \
                       RT_DEVICE_FLAG_STREAM) == RT_EOK)
    {
        /* clear line buffer before switch to new device */
        memset(shell->line, 0, sizeof(shell->line));
        shell->line_curpos = shell->line_position = 0;
	}
}`

就应该可以工作了吧。

BernardXiong, 2017-11-20:
STM32f10x_HAL 代码，初始化部分估计有问题。

armink, 2018-02-24:
增加一个宏判断下就行了

```C
#if defined(RT_USING_FINSH) && !defined(RT_USING_POSIX)
    finsh_set_device(RT_CONSOLE_DEVICE_NAME);
#endif
```

closed 2020-06-24 by BernardXiong