RT-Thread/rt-thread #1021: bsp\imxrt1050-evk编译错误

XieyangLy, 2017-11-21:
下面是env的输出信息

Active code page: 437

E:\RT-thread\env>cd ..\rt-thread-3.0.0\bsp\imxrt1052-evk

E:\RT-thread\rt-thread-3.0.0\bsp\imxrt1052-evk>ls
BuildLog.log  bsp                  project.ewp      sdram_init.jlinkscript
KConfig       drivers              project.ewt      sdram_mpu_init.ini
Libraries     imxrt1052_sdram.icf  project.eww      sdram_mpu_init.mac
SConscript    imxrt1052_sdram.ld   project.uvprojx  template.ewp
SConstruct    imxrt1052_sdram.sct  rtconfig.h       template.uvprojx
applications  project.ewd          rtconfig.py

E:\RT-thread\rt-thread-3.0.0\bsp\imxrt1052-evk>scons
scons: Reading SConscript files ...
E:\RT-thread\rt-thread-3.0.0
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build\applications\device_test.o
CC build\applications\main.o
In file included from Libraries/MIMXRT1052.h:287:0,
                 from Libraries/fsl_device_registers.h:44,
                 from Libraries\drivers/fsl_common.h:44,
                 from drivers/board.h:19,
                 from applications\main.c:33:
Libraries\CMSIS\Include/core_cm7.h:103:0: warning: "__FPU_USED" redefined
       #define __FPU_USED       1U
 ^
<command-line>:0:0: note: this is the location of the previous definition
applications\main.c:39:13: warning: 'dump_clock' defined but not used [-Wunused-function]
 static void dump_clock(void)
             ^
CC build\applications\mem_test.o
applications\mem_test.c: In function 'mem_test':
applications\mem_test.c:81:13: warning: value computed is not used [-Wunused-value]
             *p_uint32_t++;
             ^
CC build\applications\sdio_test.o
applications\sdio_test.c: In function 'write_sd__':
applications\sdio_test.c:84:14: warning: variable 'result' set but not used [-Wunused-but-set-variable]
     rt_err_t result;
              ^
CC build\drivers\board.o
In file included from Libraries/MIMXRT1052.h:287:0,
                 from Libraries/fsl_device_registers.h:44,
                 from Libraries\drivers/fsl_common.h:44,
                 from drivers\board.h:19,
                 from drivers\board.c:18:
Libraries\CMSIS\Include/core_cm7.h:103:0: warning: "__FPU_USED" redefined
       #define __FPU_USED       1U
 ^
<command-line>:0:0: note: this is the location of the previous definition
CC build\drivers\drv_eth.o
In file included from Libraries/MIMXRT1052.h:287:0,
                 from Libraries/fsl_device_registers.h:44,
                 from Libraries\drivers/fsl_common.h:44,
                 from drivers\board.h:19,
                 from drivers\drv_eth.c:15:
Libraries\CMSIS\Include/core_cm7.h:103:0: warning: "__FPU_USED" redefined
       #define __FPU_USED       1U
 ^
<command-line>:0:0: note: this is the location of the previous definition
CC build\drivers\drv_sdio.o
In file included from Libraries/MIMXRT1052.h:287:0,
                 from Libraries/fsl_device_registers.h:44,
                 from Libraries\drivers/fsl_common.h:44,
                 from drivers/board.h:19,
                 from drivers\drv_sdio.c:29:
Libraries\CMSIS\Include/core_cm7.h:103:0: warning: "__FPU_USED" redefined
       #define __FPU_USED       1U
 ^
<command-line>:0:0: note: this is the location of the previous definition
CC build\drivers\fsl_phy.o
In file included from Libraries/MIMXRT1052.h:287:0,
                 from Libraries/fsl_device_registers.h:44,
                 from Libraries\drivers/fsl_common.h:44,
                 from Libraries\drivers/fsl_enet.h:33,
                 from drivers\fsl_phy.h:33,
                 from drivers\fsl_phy.c:31:
Libraries\CMSIS\Include/core_cm7.h:103:0: warning: "__FPU_USED" redefined
       #define __FPU_USED       1U
 ^
<command-line>:0:0: note: this is the location of the previous definition
CC build\drivers\usart.o
In file included from Libraries/MIMXRT1052.h:287:0,
                 from Libraries/fsl_device_registers.h:44,
                 from Libraries\drivers/fsl_common.h:44,
                 from drivers\usart.c:17:
Libraries\CMSIS\Include/core_cm7.h:103:0: warning: "__FPU_USED" redefined
       #define __FPU_USED       1U
 ^
<command-line>:0:0: note: this is the location of the previous definition
CC build\kernel\components\dfs\filesystems\devfs\devfs.o
CC build\kernel\components\dfs\filesystems\elmfat\dfs_elm.o
CC build\kernel\components\dfs\filesystems\elmfat\ff.o
CC build\kernel\components\dfs\filesystems\net\dfs_net.o
CC build\kernel\components\dfs\filesystems\net\net_netdb.o
CC build\kernel\components\dfs\filesystems\net\net_sockets.o
CC build\kernel\components\dfs\src\dfs.o
CC build\kernel\components\dfs\src\dfs_file.o
CC build\kernel\components\dfs\src\dfs_fs.o
CC build\kernel\components\dfs\src\dfs_posix.o
CC build\kernel\components\dfs\src\poll.o
CC build\kernel\components\dfs\src\select.o
In file included from E:\RT-thread\rt-thread-3.0.0\components\dfs\src\select.c:29:0:
E:\RT-thread\rt-thread-3.0.0\components\dfs\include/dfs_select.h:10:22: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout);
                      ^
E:\RT-thread\rt-thread-3.0.0\components\dfs\include/dfs_select.h:10:39: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout);
                                       ^
E:\RT-thread\rt-thread-3.0.0\components\dfs\include/dfs_select.h:10:57: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout);
                                                         ^
E:\RT-thread\rt-thread-3.0.0\components\dfs\src\select.c:31:22: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout)
                      ^
E:\RT-thread\rt-thread-3.0.0\components\dfs\src\select.c:31:39: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout)
                                       ^
E:\RT-thread\rt-thread-3.0.0\components\dfs\src\select.c:31:57: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout)
                                                         ^
scons: *** [build\kernel\components\dfs\src\select.o] Error 1
scons: building terminated because of errors.

E:\RT-thread\rt-thread-3.0.0\bsp\imxrt1052-evk>

TanekLiang, 2017-12-08:
__FPU_USED的警告我修复了。
编译错误的问题， 你可以试试现在可以编译了吗？

XieyangLy, 2017-12-31:
最新的可以编译通过。好多天以前就测试过了，但是忘记回复了

closed 2017-12-31 by XieyangLy

bjrobertshen, 2018-05-18:
Just download the latest RT_THREAD with micropyhthon and using scons to build with gcc, I med the same issue as above:
robert@robert-desktop:~/mydisk/code/rt-thread/bsp/imxrt1052-evk$ scons
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/Libraries/drivers/fsl_adc.o
CC build/Libraries/drivers/fsl_adc_etc.o
CC build/Libraries/drivers/fsl_aipstz.o
CC build/Libraries/drivers/fsl_aoi.o
CC build/Libraries/drivers/fsl_bee.o
CC build/Libraries/drivers/fsl_cache.o
CC build/Libraries/drivers/fsl_clock.o
CC build/Libraries/drivers/fsl_cmp.o
CC build/Libraries/drivers/fsl_common.o
CC build/Libraries/drivers/fsl_csi.o
CC build/Libraries/drivers/fsl_dcdc.o
CC build/Libraries/drivers/fsl_dcp.o
CC build/Libraries/drivers/fsl_dmamux.o
CC build/Libraries/drivers/fsl_edma.o
CC build/Libraries/drivers/fsl_elcdif.o
CC build/Libraries/drivers/fsl_enc.o
CC build/Libraries/drivers/fsl_enet.o
CC build/Libraries/drivers/fsl_ewm.o
CC build/Libraries/drivers/fsl_flexcan.o
CC build/Libraries/drivers/fsl_flexio.o
CC build/Libraries/drivers/fsl_flexio_i2c_master.o
CC build/Libraries/drivers/fsl_flexio_i2s.o
CC build/Libraries/drivers/fsl_flexio_i2s_edma.o
CC build/Libraries/drivers/fsl_flexio_spi.o
CC build/Libraries/drivers/fsl_flexio_spi_edma.o
CC build/Libraries/drivers/fsl_flexio_uart.o
CC build/Libraries/drivers/fsl_flexio_uart_edma.o
CC build/Libraries/drivers/fsl_flexram.o
CC build/Libraries/drivers/fsl_flexspi.o
CC build/Libraries/drivers/fsl_gpc.o
CC build/Libraries/drivers/fsl_gpio.o
CC build/Libraries/drivers/fsl_gpt.o
CC build/Libraries/drivers/fsl_kpp.o
CC build/Libraries/drivers/fsl_lpi2c.o
CC build/Libraries/drivers/fsl_lpi2c_edma.o
CC build/Libraries/drivers/fsl_lpspi.o
CC build/Libraries/drivers/fsl_lpspi_edma.o
CC build/Libraries/drivers/fsl_lpuart.o
CC build/Libraries/drivers/fsl_lpuart_edma.o
CC build/Libraries/drivers/fsl_pit.o
CC build/Libraries/drivers/fsl_pmu.o
CC build/Libraries/drivers/fsl_pwm.o
CC build/Libraries/drivers/fsl_pxp.o
CC build/Libraries/drivers/fsl_qtmr.o
CC build/Libraries/drivers/fsl_rtwdog.o
CC build/Libraries/drivers/fsl_sai.o
CC build/Libraries/drivers/fsl_sai_edma.o
CC build/Libraries/drivers/fsl_semc.o
CC build/Libraries/drivers/fsl_snvs_hp.o
CC build/Libraries/drivers/fsl_snvs_lp.o
CC build/Libraries/drivers/fsl_spdif.o
CC build/Libraries/drivers/fsl_spdif_edma.o
CC build/Libraries/drivers/fsl_src.o
CC build/Libraries/drivers/fsl_trng.o
CC build/Libraries/drivers/fsl_tsc.o
CC build/Libraries/drivers/fsl_usdhc.o
CC build/Libraries/drivers/fsl_wdog.o
CC build/Libraries/drivers/fsl_xbara.o
CC build/Libraries/drivers/fsl_xbarb.o
CC build/applications/device_test.o
CC build/applications/main.o
CC build/applications/mem_dump.o
CC build/applications/mem_test.o
CC build/drivers/board.o
CC build/drivers/drv_cache.o
CC build/drivers/drv_eth.o
CC build/drivers/drv_i2c.o
CC build/drivers/drv_lcd.o
CC build/drivers/drv_pin.o
drivers/drv_pin.c: In function 'rt1052_get_irqnum':
drivers/drv_pin.c:423:12: warning: 'irq_num' may be used uninitialized in this function [-Wmaybe-uninitialized]
     return irq_num;
               ^~~~~~~
CC build/drivers/drv_rtc.o
CC build/drivers/drv_sdio.o
CC build/drivers/drv_sdram.o
CC build/drivers/drv_spi_bus.o
CC build/drivers/drv_uart.o
CC build/drivers/fsl_phy.o
CC build/kernel/components/dfs/filesystems/devfs/devfs.o
CC build/kernel/components/dfs/filesystems/elmfat/dfs_elm.o
CC build/kernel/components/dfs/filesystems/elmfat/ff.o
CC build/kernel/components/dfs/filesystems/elmfat/option/ccsbcs.o
CC build/kernel/components/dfs/filesystems/net/dfs_net.o
CC build/kernel/components/dfs/filesystems/net/net_netdb.o
CC build/kernel/components/dfs/filesystems/net/net_sockets.o
CC build/kernel/components/dfs/src/dfs.o
CC build/kernel/components/dfs/src/dfs_file.o
CC build/kernel/components/dfs/src/dfs_fs.o
CC build/kernel/components/dfs/src/dfs_posix.o
CC build/kernel/components/dfs/src/poll.o
CC build/kernel/components/dfs/src/select.o
In file included from /home/robert/mydisk/code/rt-thread/components/dfs/src/select.c:29:0:
/home/robert/mydisk/code/rt-thread/components/dfs/include/dfs_select.h:10:22: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout);
                             ^~~~~~
/home/robert/mydisk/code/rt-thread/components/dfs/include/dfs_select.h:10:39: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout);
                                                       ^~~~~~
/home/robert/mydisk/code/rt-thread/components/dfs/include/dfs_select.h:10:57: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout);
                                                         ^~~~~~
/home/robert/mydisk/code/rt-thread/components/dfs/src/select.c:31:22: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout)
                             ^~~~~~
/home/robert/mydisk/code/rt-thread/components/dfs/src/select.c:31:39: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout)
                                                       ^~~~~~
/home/robert/mydisk/code/rt-thread/components/dfs/src/select.c:31:57: error: unknown type name 'fd_set'
 int select(int nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout)
                                                         ^~~~~~
scons: *** [build/kernel/components/dfs/src/select.o] Error 1
scons: building terminated because of errors.
robert@robert-desktop:~/mydisk/code/rt-thread/bsp/imxrt1052-evk$