RT-Thread/rt-thread #1022: 请完善cmsis os接口

biansf2001, 2017-11-21:
因为要使用rtt+emwin，在emwin532版本里，GUI_X_OS.c是和os的接口，在rtconfig.h里打开#define RT_USING_CMSIS_OS后，编译出了几条错误，如下
cmsis_rtthread.c  
Error[Pe167]: argument of type "os_pthread" is incompatible with parameter of type "void (*)(void *)" \RTThread\rt-thread-v2.1.x\components\CMSIS\RTOS\cmsis_rtthread.c 33 
Warning[Pe188]: enumerated type mixed with another type \RTThread\rt-thread-v2.1.x\components\CMSIS\RTOS\cmsis_rtthread.c 100 
Warning[Pe940]: missing return statement at end of non-void function "osWait" \RTThread\rt-thread-v2.1.x\components\CMSIS\RTOS\cmsis_rtthread.c 134 
Warning[Pe550]: variable "result" was set but never used \RTThread\rt-thread-v2.1.x\components\CMSIS\RTOS\cmsis_rtthread.c 123 
Error[Pe167]: argument of type "os_ptimer" is incompatible with parameter of type "void (*)(void *)" \RTThread\rt-thread-v2.1.x\components\CMSIS\RTOS\cmsis_rtthread.c 148 
Error while running C/C++ Compiler

BernardXiong, 2017-11-21:
可以考虑发PR上来。

mysterywolf, 2021-02-06:
请使用此软件包：http://packages.rt-thread.org/detail.html?package=CMSIS 该软件包包含了CMSIS-OS兼容接口

closed 2021-02-06 by mysterywolf