RT-Thread/rt-thread #1045: stm32f429-apollo bsp 在开发板上无法使用 keil版本，27日更新的

KevinZJX, 2017-11-28:
stm32f429-apollo板由于没有分散加载文件，导致无法运行
运行结果为：
initialize rti_board_start:0 done
initialize stm32_hw_usart_init:0 done
initialize rt_sdram_hw_init:0 done
initialize rt_hw_rtc_init

望更新

uestczyh222, 2017-11-28:
尝试下F4xx-HAL bsp

uestczyh222, 2017-11-28:
TanekLiang 2017-11-28 14:46:03
嗯嗯， 有不少mdk是缺少独立的分散加载文件的， 只能在ide里面编译。
14:46:34
TanekLiang 2017-11-28 14:46:34
可以在mdk里面生成一个复制出来使用。
TanekLiang 2017-11-28 14:46:44
然后提交pr。

KevinZJX, 2017-11-28:
还是不正确，我对源码进行了修改，将一些宏定义修改了，但还是不行
#ifdef __CC_ARM
extern int Image$$RW_IRAM1$$ZI$$Limit;
//#define STM32_SRAM_BEGIN    (&Image$$RW_IRAM1$$ZI$$Limit)
#define STM32_SRAM_BEGIN	0x20000000
可能源码中还有更深的替换

uestczyh222, 2017-11-28:
这部分你参考下F4xx-HAL 是怎么写的

KevinZJX, 2017-11-28:
好的

KevinZJX, 2017-11-28:
我发现并不是那个文件导致无法运行的，在Linker里直接选取系统默认的也可以运行，真正出问题的地方是是在初始化RTC时会死在
while(__HAL_RCC_GET_FLAG(RCC_FLAG_LSERDY) == RESET)
      {
        if((HAL_GetTick() - tickstart ) > RCC_LSE_TIMEOUT_VALUE)
        {
          return HAL_TIMEOUT;
        }       
      }
这个里面，无法退出，我取消了RTC的初始化就可以了正常运行了
int  rt_hw_rtc_init(void)
{
//    RTC_Init();
//    /* register rtc device */
//		rt_hw_rtc_register(&rtc, RT_RTC_NAME, 0);
    return RT_EOK;
}

KevinZJX, 2017-11-28:
finsh />list()
--Function List:
list_mem         -- list memory usage information
ls               -- list directory contents
rm               -- remove files or directories
cat              -- print file
copy             -- copy file or dir
mkfs             -- make a file system
df               -- get disk free
mkdir            -- create a directory
cd               -- change current working directory
list_date        -- show date and time.
set_date         -- set date. e.g: set_date(2010,2,28)
set_time         -- set time. e.g: set_time(23,59,59)
hello            -- say hello world
version          -- show RT-Thread version information
list_thread      -- list thread
list_sem         -- list semaphone in system
list_event       -- list event in system
list_mutex       -- list mutex in system
list_mailbox     -- list mail box in system
list_msgqueue    -- list message queue in system
list_memheap     -- list memory heap in system
list_mempool     -- list memory pool in system
list_timer       -- list timer in system
list_device      -- list device in system
list             -- list all symbol in system
set_if           -- set network interface address
set_dns          -- set DNS server address
list_if          -- list network interface information
list_tcps        -- list all of tcp connections
--Variable List:
dummy            -- dummy variable for finsh
        0, 0x00000000
finsh />

KevinZJX, 2017-11-28:
initialize rti_board_start:0 done
initialize stm32_hw_usart_init:0 done
initialize rt_sdram_hw_init:0 done
initialize rt_hw_rtc_init:0 done
initialize mpu_init:0 done
initialize rt_hw_sdcard_init:0 done
initialize nand_MT29F4G08_hw_init
NAND ID: 0xDC909556
:0 done

 \ | /
- RT -     Thread Operating System
 / | \     3.0.0 build Nov 28 2017
 2006 - 2017 Copyright by rt-thread team
do components intialization.
initialize rti_board_end:0 done
initialize dfs_init:0 done
initialize eth_system_device_init:0 done
initialize rt_hw_stm32_eth_init:0 done
initialize stm32f4_i2c_init:0 done
initialize elm_init:0 done
initialize rt_i2c_core_init:0 done
initialize libc_system_init:0 done
initialize lwip_system_initlwIP-2.0.2 initialized!
:0 done
initialize finsh_system_init:0 done
File System initialzation failed!
spi flash mount to /spi failed!
finsh />

TanekLiang, 2017-11-30:
嗯，应该是hal的 delay被重载的缘故，导致没法超时。(｡•́︿•̀｡)

KevinZJX, 2017-12-01:
我试着修复一下看，然后提交

mysterywolf, 2021-02-06:
2021-2-6 Keil编译无误 问题已经得到解决

closed 2021-02-06 by mysterywolf