RT-Thread/rt-thread #1121: rt-thread的USB 驱动框架疑问

xujuncz, 2017-12-20:
我在移植stm32f407的USB HOST驱动时，发现rt-thread的USB驱动框架与USB协议有一些差别的，特别是与Windows下的枚举流程是有区别的。
1、在获取设备描述符的头信息后，windows下是需要再次复位USB设备的。最好能在驱动框架中实现一个USB端口复位函数。
2、rt-thread的usb框架应该是基于OHCI框架写的，但里面没有把根集线器定义出来，根集线器是空的，实际这个根集线器与具体硬件相关性不大的，应该可以在驱动框架中实现出来
rt_uint8_t root_hub_desc[] = 
{
    0x17,
    0x29,
    1,
    0x00,
	0x00,
    5,
    10,
    0,
    0,
    0,
    0,
    0,
    0,
    0,
    0,
    0xff,
    0xff,
    0xff,
    0xff,
    0xff,
    0xff,
    0xff,
    0xff,
};
struct uhub* roothub_init(void)
{
	struct uhub* rh0 = &hub0;
	rt_memset((rt_uint8_t*) rh0,0,sizeof(struct uhub));
	int i = 0;
	
	//roothub desc initial
	rt_memcpy(&rh0->hub_desc,root_hub_desc,sizeof(struct uhub_descriptor));
		
	rh0->num_ports = 1;
	
	for(i = 0; i < USB_HUB_PORT_NUM; i++)
	{
		//PORT ENABLE,NO CONNECT, PWR ENABLE
		rh0->port_status[i] = PORT_PES | PORT_PPS;
	}
	rh0->pipe_in = RT_NULL;
	rh0->is_roothub = 1;
	//
	rt_memset(&rh0->buffer[0],0,8);

	rh0->self 	= RT_NULL;
	rh0->hcd 	= RT_NULL;
	return rh0;
}

uestczyh222, 2017-12-20:
你好 有实现的 在hub.c里面。除了描述符以外都是有实现的

uestczyh222, 2017-12-20:
另外 是有端口复位函数的具体可以看F4xx-HAL里面的移植

uestczyh222, 2017-12-20:
控制器通过
void rt_usbh_root_hub_connect_handler(struct uhcd *hcd, rt_uint8_t port, rt_bool_t isHS)
void rt_usbh_root_hub_disconnect_handler(struct uhcd *hcd, rt_uint8_t port)
实现通知roothub端口的断开与连接
协议栈通过
struct uhcd_ops
{
    rt_err_t    (*reset_port)   (rt_uint8_t port);
    int         (*pipe_xfer)    (upipe_t pipe, rt_uint8_t token, void* buffer, int nbytes, int timeout);
    rt_err_t    (*open_pipe)    (upipe_t pipe);
    rt_err_t    (*close_pipe)   (upipe_t pipe);  
};
中的reset_port复位端口

uestczyh222, 2017-12-20:
另外大兄弟有空的话可以帮忙修复一下hub的驱动

uestczyh222, 2017-12-20:
话说你是不是用的老版本的代码 最新的host框架经过了简单修复 已经可以使用了 前两天刚修的

closed 2018-01-23 by uestczyh222