RT-Thread/rt-thread #1127: STM32F4xxx: MDK5不能编译，但是 Scons可以

sczh01, 2017-12-22:
复现路径:
1. 打开 #define RT_USING_USB_HOST
2. 在scons下 scons --target=mdk5 -s
3. 直接在scons 用scons编译可以通过。
LINK rtthread-stm32f4xx.axf
fromelf --bin rtthread-stm32f4xx.axf --output rtthread.bin
fromelf -z rtthread-stm32f4xx.axf

========================================================================

** Object/Image Component Sizes

      Code (inc. data)   RO Data    RW Data    ZI Data      Debug   Object Name

     60768       6622       7364        592      12048     998058   rtthread-stm32f4xx.axf (uncompressed)
     60768       6622       7364        128      12048     998058   rtthread-stm32f4xx.axf (compressed)
     60768       6622       7364        128          0          0   ROM Totals for rtthread-stm32f4xx.axf
scons: done building targets.
4.用MDK5.23打开编译： 出现如下6个编译错误，看错误同头文件处理相关
Libraries\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_pwr_ex.c(399): error:  #20: identifier "PWR_CR_MRLVDS_Pos" is undefined
    *(
_IO uint32_t *) 
R_MRLVDS_BB = (uint32_t)ENABLE;
Libraries\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_pwr_ex.c(410): error:  #20: identifier "PWR_CR_MRLVDS_Pos" is undefined
    *(
_IO uint32_t *) 
R_MRLVDS_BB = (uint32_t)DISABLE;
Libraries\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_pwr_ex.c(421): error:  #20: identifier "PWR_CR_LPLVDS_Pos" is undefined
    *(
_IO uint32_t *) 
R_LPLVDS_BB = (uint32_t)ENABLE;
Libraries\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_pwr_ex.c(432): error:  #20: identifier "PWR_CR_LPLVDS_Pos" is undefined
Libraries\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc_ex.c(2666): error:  #136: struct "<unnamed>"  has no field "DCKCFGR"
    if ((
CC->DCKCFGR & RCC_DCKCFGR_TIMPRE) == RESET)
Libraries\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_rcc_ex.c(2666): error:  #20: identifier "RCC_DCKCFGR_TIMPRE" is undefined
    if ((

uestczyh222, 2017-12-22:
你是不是换了芯片没有在mdk工程里重新选芯片包

sczh01, 2017-12-23:
确实是这个问题

sczh01, 2017-12-23:
还有一个bug,我直接在这里说吧，当我同时打开HOST和DEVICE后 会出现如下连接错误。
是否不能同时打开呢？

linking...
.\build\rtthread-stm32f4xx.axf: Error: L6200E: Symbol OTG_FS_IRQHandler multiply defined (by drv_usbh.o and drv_usbd.o).
Not enough information to list image symbols.
Not enough information to list the image map.
Finished: 2 information, 0 warning and 1 error messages.
".\build\rtthread-stm32f4xx.axf" - 1 Error(s), 1 Warning(s).

closed 2017-12-23 by sczh01