RT-Thread/rt-thread #1133: Simulator下不能正常生成 VS2012的工程文件  VS2005可以

sczh01, 2017-12-23:
I:\tz\rt-thread\bsp\simulator>scons --target=vs2012 -s

scons: warning: No version of Visual Studio compiler found - C/C++ compilers most likely not set correctly
File "I:\tz\rt-thread\bsp\simulator\SConstruct", line 13, in <module>

scons: warning: No version of Visual Studio compiler found - C/C++ compilers most likely not set correctly
File "I:\tz\rt-thread\bsp\simulator\SConstruct", line 15, in <module>
include /src/include/posix
AttributeError: 'module' object has no attribute 'source_list':
  File "I:\tz\rt-thread\bsp\simulator\SConstruct", line 118:
    EndBuilding(TARGET, program)
  File "I:\tz\rt-thread\tools\building.py", line 699:
    VS2012Project('project.vcxproj', Projects, program)
  File "I:\tz\rt-thread\tools\vs2012.py", line 168:
    VS_add_HeadFiles(program, elem, project_path)
  File "I:\tz\rt-thread\tools\vs2012.py", line 137:
    for f in building.source_list:

I:\tz\rt-thread\bsp\simulator>

BernardXiong, 2017-12-23:
这个应该升级到vs2015才是的，后面看看文件格式有哪些改动了

FishNeil, 2018-01-03:
将173行 building修改为utils 即可

FishNeil, 2018-01-03:
将vs2012.py文件 137行 building修改为utils 即可

closed 2021-04-17 by mysterywolf