RT-Thread/rt-thread #1156: build error in arm-gcc 7.2

mail2li, 2018-01-03:
I download and install win-32bit arm-gcc "7-2017-q4-major" from https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads
modify RTT_EXEC_PATH item in console_64.bat, point to arm-gcc-7 path.
run in env shell, use default configuation of bsp\stm32f429-apollo. build fail:

```
D:\works\rt-thread\bsp\stm32f429-apollo>scons
scons: Reading SConscript files ...

scons: warning: No version of Visual Studio compiler found - C/C++ compilers most likely not set correctly
File "D:\works\rt-thread\bsp\stm32f429-apollo\SConstruct", line 32, in <module>
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build\applications\application.o
In file included from D:\works\rt-thread\include/libc/libc_signal.h:73:0,
                 from D:\works\rt-thread\include/rtlibc.h:35,
                 from D:\works\rt-thread\include/rtdef.h:1037,
                 from D:\works\rt-thread\include/rtthread.h:37,
                 from drivers/drv_sdram.h:24,
                 from drivers/board.h:20,
                 from applications\application.c:16:
d:\gnu tools arm embedded\7.2 2017q4\arm-none-eabi\include\sys\signal.h:42:7: error: redefinition of 'union sigval'
 union sigval {
       ^~~~~~
In file included from D:\works\rt-thread\include/rtlibc.h:35:0,
                 from D:\works\rt-thread\include/rtdef.h:1037,
                 from D:\works\rt-thread\include/rtthread.h:37,
                 from drivers/drv_sdram.h:24,
                 from drivers/board.h:20,
                 from applications\application.c:16:
D:\works\rt-thread\include/libc/libc_signal.h:37:7: note: originally defined here
 union sigval
       ^~~~~~
In file included from D:\works\rt-thread\include/libc/libc_signal.h:73:0,
                 from D:\works\rt-thread\include/rtlibc.h:35,
                 from D:\works\rt-thread\include/rtdef.h:1037,
                 from D:\works\rt-thread\include/rtthread.h:37,
                 from drivers/drv_sdram.h:24,
                 from drivers/board.h:20,
                 from applications\application.c:16:
d:\gnu tools arm embedded\7.2 2017q4\arm-none-eabi\include\sys\signal.h:47:8: error: redefinition of 'struct sigevent'
 struct sigevent {
        ^~~~~~~~
In file included from D:\works\rt-thread\include/rtlibc.h:35:0,
                 from D:\works\rt-thread\include/rtdef.h:1037,
                 from D:\works\rt-thread\include/rtthread.h:37,
                 from drivers/drv_sdram.h:24,
                 from drivers/board.h:20,
                 from applications\application.c:16:
D:\works\rt-thread\include/libc/libc_signal.h:43:8: note: originally defined here
 struct sigevent
        ^~~~~~~~
In file included from D:\works\rt-thread\include/libc/libc_signal.h:73:0,
                 from D:\works\rt-thread\include/rtlibc.h:35,
                 from D:\works\rt-thread\include/rtdef.h:1037,
                 from D:\works\rt-thread\include/rtthread.h:37,
                 from drivers/drv_sdram.h:24,
                 from drivers/board.h:20,
                 from applications\application.c:16:
d:\gnu tools arm embedded\7.2 2017q4\arm-none-eabi\include\sys\signal.h:72:3: error: conflicting types for 'siginfo_t'
 } siginfo_t;
   ^~~~~~~~~
In file included from D:\works\rt-thread\include/rtlibc.h:35:0,
                 from D:\works\rt-thread\include/rtdef.h:1037,
                 from D:\works\rt-thread\include/rtthread.h:37,
                 from drivers/drv_sdram.h:24,
                 from drivers/board.h:20,
                 from applications\application.c:16:
D:\works\rt-thread\include/libc/libc_signal.h:61:24: note: previous declaration of 'siginfo_t' was here
 typedef struct siginfo siginfo_t;
                        ^~~~~~~~~
scons: *** [build\applications\application.o] Error 1
scons: building terminated because of errors.

D:\works\rt-thread\bsp\stm32f429-apollo>arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=d:/gnu\ tools\ arm\ embedded/7.2\ 2017q4/bin/../lib/gcc/arm-none-eabi/7.2.1/lto-wrapper.exe
Target: arm-none-eabi
Configured with: /tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/src/gcc/configure --build=x86_64-linux-gnu --host=i686-w64-mingw32 --target=arm-none-eabi --prefix=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/install-mingw --libexecdir=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/install-mingw/lib --infodir=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/install-mingw/share/doc/gcc-arm-none-eabi/info --mandir=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/install-mingw/share/doc/gcc-arm-none-eabi/man --htmldir=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/install-mingw/share/doc/gcc-arm-none-eabi/html --pdfdir=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/install-mingw/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-mingw-wildcard --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-headers=yes --with-newlib --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/install-mingw/arm-none-eabi --with-libiconv-prefix=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/build-mingw/host-libs/usr --with-gmp=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/build-mingw/host-libs/usr
--with-mpfr=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/build-mingw/host-libs/usr --with-mpc=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/build-mingw/host-libs/usr --with-isl=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/build-mingw/host-libs/usr --with-libelf=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/build-mingw/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Tools for Arm Embedded Processors 7-2017-q4-major' --with-multilib-list=rmprofile
Thread model: single
gcc version 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204] (GNU Tools for Arm Embedded Processors 7-2017-q4-major)
```

BernardXiong, 2018-01-03:
Please use the toolchains inside env. The different version of toolchains has different libc header files.

Pillar1989, 2018-01-12:
@BernardXiong  any documents about set libc header files? I have the same issue too.

BernardXiong, 2018-01-12:
You can follow the travis-ci building script:
https://github.com/RT-Thread/rt-thread/blob/master/.travis.yml

Pillar1989, 2018-01-12:
.travis.yml use the gcc5,  doest rt_thread only works on gcc5 ?
That's my gcc version.
```
baozhu@server:/tmp/rt-thread/bsp/qemu-vexpress-a9$ /tmp/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=/tmp/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/tmp/gcc-arm-none-eabi-7-2017-q4-major/bin/../lib/gcc/arm-none-eabi/7.2.1/lto-wrapper
Target: arm-none-eabi
Configured with: /tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/src/gcc/configure --target=arm-none-eabi --prefix=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/install-native --libexecdir=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/install-native/lib --infodir=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/install-native/share/doc/gcc-arm-none-eabi/info --mandir=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/install-native/share/doc/gcc-arm-none-eabi/man --htmldir=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/install-native/share/doc/gcc-arm-none-eabi/html --pdfdir=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/install-native/share/doc/gcc-arm-none-eabi/pdf --enable-languages=c,c++ --enable-plugins --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-shared --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/install-native/arm-none-eabi --build=x86_64-linux-gnu --host=x86_64-linux-gnu --with-gmp=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/build-native/host-libs/usr --with-mpfr=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/build-native/host-libs/usr --with-mpc=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/build-native/host-libs/usr --with-isl=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/build-native/host-libs/usr --with-libelf=/tmp/jenkins/jenkins-GCC-7-build_toolchain_docker-633_20171130_1512067137/build-native/host-libs/usr --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --with-pkgversion='GNU Tools for Arm Embedded Processors 7-2017-q4-major' --with-multilib-list=rmprofile
Thread model: single
gcc version 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204] (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 

```

BernardXiong, 2018-01-12:
If not gcc5, the compiler related code should be porting.

Pillar1989, 2018-01-12:
Yes, GCC5 works for me . But GCC7 should be support

imgtec, 2020-03-08:
Hi, this is not related to gcc version, but platfrom. please use "Bare-Matel" platform gcc.

> `brew@revive:~/ls1b200-rtt/rt-thread/bsp$ mips-mti-elf-gcc -v
Using built-in specs.
COLLECT_GCC=mips-mti-elf-gcc
COLLECT_LTO_WRAPPER=/home/brew/opt/mips-mti-elf/2019.09-01/bin/../libexec/gcc/mips-mti-elf/7.4.0/lto-wrapper
Target: mips-mti-elf
Configured with: /scratch/overtest/root/74484/240198/shared/gcc/configure --prefix=/scratch/overtest/root/74484/240313/shared/mips-mti-elf/2019.09-01 --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --disable-libmudflap --disable-decimal-float --with-mips-plt --target=mips-mti-elf --with-newlib --with-gnu-as --with-gnu-ld --enable-languages=c,c++ --with-sysroot=/scratch/overtest/root/74484/240313/shared/mips-mti-elf/2019.09-01/mips-mti-elf --with-build-sysroot=/scratch/overtest/root/74484/240313/shared/mips-mti-elf/2019.09-01/mips-mti-elf --enable-newlib-long-time_t --enable-newlib-io-c99-formats --with-bugurl=http://mips.com/mips-sdk-support/ --with-pkgversion='Codescape GNU Tools 2019.09-01 for MIPS MTI Bare Metal'
Thread model: single
gcc version 7.4.0 (Codescape GNU Tools 2019.09-01 for MIPS MTI Bare Metal)`

mysterywolf, 2022-01-21:
现在用gcc10编译 啥毛病没有

closed 2022-01-21 by mysterywolf