RT-Thread/rt-thread #1183: bsp tm4c129x, gcc compiler error when USE RT_USING_LIBC or RT_USING_PTHREADS

mhl8002, 2018-01-22:


BernardXiong, 2018-01-22:
minilibc是给gcc使用的。在gcc工具链状态下：当不是用libc时，minilibc会被自动启用。当使用libc时，newlib会被使用。

mhl8002, 2018-01-22:
谢谢, 
开启 RT_USING_LIBC之后,编译出现以下错误,帮忙看看是什么问题, 谢谢.

scons: building associated VariantDir targets: build
CC build/applications/application.o
CC build/applications/board.o
<command-line>:0:13: error: conflicting types for 'int32_t'
In file included from /usr/include/newlib/sys/types.h:63:0,
                 from /usr/include/newlib/time.h:27,
                 from /usr/include/newlib/sys/stat.h:9,
                 from /home/ocean/prj/ble/rt-thread/include/libc/libc_stat.h:8,
                 from /home/ocean/prj/ble/rt-thread/include/rtlibc.h:29,
                 from /home/ocean/prj/ble/rt-thread/include/rtdef.h:1037,
                 from /home/ocean/prj/ble/rt-thread/include/rtthread.h:37,
                 from /home/ocean/prj/ble/rt-thread/include/rthw.h:32,
                 from applications/board.c:16:
/usr/include/newlib/sys/_stdint.h:31:19: note: previous declaration of 'int32_t' was here
 typedef __int32_t int32_t ;
                   ^
<command-line>:0:14: error: conflicting types for 'uint32_t'
In file included from /usr/include/newlib/sys/types.h:63:0,
                 from /usr/include/newlib/time.h:27,
                 from /usr/include/newlib/sys/stat.h:9,
                 from /home/ocean/prj/ble/rt-thread/include/libc/libc_stat.h:8,
                 from /home/ocean/prj/ble/rt-thread/include/rtlibc.h:29,
                 from /home/ocean/prj/ble/rt-thread/include/rtdef.h:1037,
                 from /home/ocean/prj/ble/rt-thread/include/rtthread.h:37,
                 from /home/ocean/prj/ble/rt-thread/include/rthw.h:32,
                 from applications/board.c:16:
/usr/include/newlib/sys/_stdint.h:32:20: note: previous declaration of 'uint32_t' was here
 typedef __uint32_t uint32_t ;
                    ^
scons: *** [build/applications/board.o] Error 1
scons: building terminated because of errors.

gcc 版本:
 arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-none-eabi/4.9.3/lto-wrapper
Target: arm-none-eabi
........
gcc version 4.9.3 20150529 (prerelease) (15:4.9.3+svn231177-1)

BernardXiong, 2018-01-22:
这个为什么是自己和自己冲突：
/usr/include/newlib/sys/_stdint.h:31:19: note: previous declaration of 'int32_t' was here
typedef __int32_t int32_t ;
In file included from /usr/include/newlib/sys/types.h:63:0,

应该是你用的工具链有问题吧。

mhl8002, 2018-01-23:
@BernardXiong , 我找到了解决办法,请确认是不是问题所在:
在文件 bsp/tm4c129x/libraries/sconscipt去掉以下几行:
if rtconfig.CROSS_TOOL == 'gcc':
	CPPDEFINES += ['gcc'];
#	CPPDEFINES += ['uint_fast8_t=uint32_t'];       // 开启rt_using_libc,需要注释掉.
#	CPPDEFINES += ['uint_fast32_t=uint32_t'];     // 开启rt_using_libc,需要注释掉.
#	CPPDEFINES += ['int_fast8_t=int32_t'];          // 开启rt_using_libc,需要注释掉
#	CPPDEFINES += ['uint_fast16_t=uint32_t'];   // 开启rt_using_libc,需要注释掉.

BernardXiong, 2018-01-23:
好像是有些奇怪，怎么会有这些定义。如果确实因为这些引起的，请提交下PR。另外，这类如果是显著性错误，直接删除就好，不需要以注释的方式残留代码。

mhl8002, 2018-01-24:
@BernardXiong , 在使用minilibc 时, 如果没有以上定义,编译不pass.

TanekLiang, 2018-02-06:
那可以在相应的SConscript添加判断，用来区分是使用minilibc还是newlib。

例如这样：
```
if rtconfig.CROSS_TOOL == 'gcc' and not GetDepend('RT_USING_LIBC'):
	CPPDEFINES += ['gcc']
	CPPDEFINES += ['uint_fast8_t=uint32_t']
	CPPDEFINES += ['uint_fast32_t=uint32_t']
	CPPDEFINES += ['int_fast8_t=int32_t']
	CPPDEFINES += ['uint_fast16_t=uint32_t']
```

BernardXiong, 2018-02-06:
但是这个uint_fast32_t = uint32_t，貌似太奇葩了。应该根源还在于BSP，需要把BSP进行修正的。tm4c129x，是 @ArdaFu 维护的吗？

ArdaFu, 2018-03-18:
tm4c BSP的library 需要这些 xxxx_fast32_t的定义，例如在adc.c中。

BernardXiong, 2018-03-19:
这部分是ti library本身就需要的吗？还是说它可以include另外一个头文件？如果可以include另外一个头文件，可能直接在这个头文件中进行处理会更好。

ArdaFu, 2018-03-19:
这个是C99引入的定义，应该是RTT的.\components\libc\compilers\minilibc\stdint.h中定义不全导致的。gcc tool chain 中的stdint.h是包含这个定义的参见arm-none-eabi\include\stdint.h

BernardXiong, 2018-03-19:
原来是这样的历史，那么就提交到minilibc那边去吧，直接在标准头文件中加入

ArdaFu, 2018-03-19:
OK，我去弄一下

closed 2018-03-19 by ArdaFu