RT-Thread/rt-thread #1190: 关于cdc_vcom.c的 BUG 报告

refugeer, 2018-01-24:
#1160 @uestczyh222 @AubrCool 
1、
错误：
 #define **__SER_NO_LEN** RT_VCOM_SER_LEN
 #define **__SER_NO_LEN** rt_strlen("32021919830108")
修正：
 #define **_SER_NO_LEN** RT_VCOM_SER_LEN
 #define **_SER_NO_LEN** rt_strlen("32021919830108")
 原因：rt_memset(serno, 0, **_SER_NO_LEN** + 1);  
 描述：根据下面调用的提示，define的时候多了一个下划线
 处理：去除一个下划线"_"
2、
错误：
“function call is not allowed in a constant expression”
出处：
static char serno[**_SER_NO_LEN + 1**] = {'\0',};
原因：
#ifdef RT_VCOM_SER_LEN
#define _SER_NO_LEN RT_VCOM_SER_LEN
#else /*!RT_VCOM_SER_LEN*/
**#define _SER_NO_LEN rt_strlen("32021919830108")**
#endif /*RT_VCOM_SER_LEN*/
描述：如果没有define RT_VCOM_SER_LEN，将调用rt_strlen函数获取字符串长度去初始化数组，
这在keil (armcc v5.06 update4)中是违规的。
处理：
           1)   可否用sizeof代替rt_strlen ？
           2)   直接将字符串长度计算给_SER_NO_LEN，如#define _SER_NO_LEN 14

AubrCool, 2018-01-24:
确实是，应该换成sizeof

uestczyh222, 2018-01-24:
@AubrCool 你改还是我改？

uestczyh222, 2018-01-24:
搞定了

closed 2018-01-24 by uestczyh222