RT-Thread/rt-thread #1200: usb cdc_vcom疑问：

ChunfengMu, 2018-01-31:
现版本cdc_vcom程序中,设备端应用程序dma模式下操作vcom发送数据的流程为:
1:将数据buffer push到data_queue中,调用注册的dma_transmit回调函数。
2:如主机端串口设备已打开,则向tx_ringbuffer中填充数据并向vcom_tx线程告知发送事件。
3:vcom_tx线程收到事件,查看tx_ringbuffer中是否有内容。如有内容,调用rt_ringbuffer_get
从tx_ringbuffer中试图最多获取64字节数据再请求发起USB Bulk In传输。
4:等待USB Bulk In传输完成。不管传输完成与否,调用rt_hw_serial_isr将data_queue的内容pop掉
并根据data_queue的情况进行下一步操作。

在当前发送流程中,如应用程序单次发送的数据块大于64字节,流程2会将数据完全填入tx_ringbuffer,
此时tx_ringbuffer中内容将大于64字节。然而流程3仅能发起长度最大为64字节的一次USB Bulk In传输,
tx_ringbuffer内将残留部分内容并在下一次USB Bulk In传输时被附带在该次实际写入内容的前部被一同提交。

当前发送流程从总体上看不会造成数据丢失，但某些时刻可能会出现数据分段异常请况。

不知vcom dma模式设计来解决怎样的问题,是否能在流程2时根据流程3 USB Bulk In提交能力限制tx_ringbuffer填充长度,
进而保证下一次USB Bulk In提交不受上一次提交的影响？

uestczyh222, 2018-02-05:
流程3 可发起大于64字节长度的传输

uestczyh222, 2018-02-05:
此处是有些不妥 我明天处理下

ChunfengMu, 2018-02-05:
好的，那我把我的修改给提了。