RT-Thread/rt-thread #1203: 貌似RT_USING_LIBC开关的描述有误

xeonxu, 2018-02-01:
这个开关被描述为：Enable libc APIs from toolchain。然而，开启这个开关，实际使用的是rtthread自带的libc；不开启该开关，使用的是toolchain的libc。
该问题通过map文件可以确定。

BernardXiong, 2018-02-01:
没错的。打开这个选项，工具链的libc会被使能；关闭这个选项，基本上libc的功能都会被关闭。而，libc的功能是完全来自工具链的。RT-Thread原则上不提供libc功能（minilibc除外，这个也仅是一个玩具级的libc，以弥补当使用裸gcc&ld时，完成RTT自举而需要的libc API）

xeonxu, 2018-02-01:
使能该开关后，map文件如下：
 7444  __libc_init_array                                 build/kernel/components/libc/compilers/newlib/syscalls.o
 7445                                                    build/Libraries/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/startup_stm32f103xe.o
 7446                                                    /usr/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/lib/armv7-m/crt0.o
 
关闭后相应内容如下：
 6204  __libc_init_array                                 /usr/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/lib/armv7-m/libc.a(lib_a-init.o)
 6205                                                    build/Libraries/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/startup_stm32f103xe.o
 6206                                                    /usr/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/lib/armv7-m/crt0.o

BernardXiong, 2018-02-01:
这里仅仅__libc_init_array被newlib/syscalls重载了，newlib/syscalls是针对newlib需要做的移植，不代表这个就是libc。

closed 2018-02-02 by xeonxu