RT-Thread/rt-thread #1208: Ubuntu下面arm-linux-gcc编译出错

craigtao, 2018-02-05:
前辈，你好，我编译的是mini2440的bsp
我使用的编译工具信息
:/work/002_rtt/rt-thread/bsp/mini2440$ arm-linux-gcc -v
Reading specs from /work/tools/gcc-3.4.5-glibc-2.3.6/lib/gcc/arm-linux/3.4.5/specs
Configured with: /work/tools/create_crosstools/crosstool-0.43/build/arm-linux/gcc-3.4.5-glibc-2.3.6/gcc-3.4.5/configure --target=arm-linux --host=i686-host_pc-linux-gnu --prefix=/work/tools/gcc-3.4.5-glibc-2.3.6 --with-float=soft --with-headers=/work/tools/gcc-3.4.5-glibc-2.3.6/arm-linux/include --with-local-prefix=/work/tools/gcc-3.4.5-glibc-2.3.6/arm-linux --disable-nls --enable-threads=posix --enable-symvers=gnu --enable-__cxa_atexit --enable-languages=c,c++ --enable-shared --enable-c99 --enable-long-long
Thread model: posix
gcc version 3.4.5

我根据官网提供的编程文档，修改了该目录下的 rtconfig.py 文件，内容如下

、
`import os

# panel options
# 'PNL_A70','PNL_N35', 'PNL_T35' , 'PNL_X35'
RT_USING_LCD_TYPE = 'PNL_T35'

# toolchains options
ARCH     = 'arm'
CPU      = 's3c24x0'
TextBase = '0x30000000'

CROSS_TOOL 	= 'arm-linux-gcc'

if os.getenv('RTT_CC'):
    CROSS_TOOL = os.getenv('RTT_CC')

if  CROSS_TOOL == 'arm-linux-gcc':
    PLATFORM 	= 'gcc'
    EXEC_PATH 	= '/work/tools/gcc-3.4.5-glibc-2.3.6/bin/'
elif CROSS_TOOL == 'keil':
    PLATFORM 	= 'armcc'
    EXEC_PATH 	= 'C:/Keil'
elif CROSS_TOOL == 'iar':
    print '================ERROR============================'
    print 'Not support iar yet!'
    print '================================================='
    exit(0)

if os.getenv('RTT_EXEC_PATH'):
	EXEC_PATH = os.getenv('RTT_EXEC_PATH')

BUILD = 'debug'

if PLATFORM == 'gcc':
    # toolchains
    PREFIX = 'arm-linux-'
    CC = PREFIX + 'gcc'
    CXX = PREFIX + 'g++'
    AS = PREFIX + 'gcc'
    AR = PREFIX + 'ar'
    LINK = PREFIX + 'gcc'
    TARGET_EXT = 'axf'
    SIZE = PREFIX + 'size'
    OBJDUMP = PREFIX + 'objdump'
    OBJCPY = PREFIX + 'objcopy'

    DEVICE = ' -mcpu=arm920t'
    CFLAGS = DEVICE
    AFLAGS = ' -c' + DEVICE + ' -x assembler-with-cpp' + ' -DTEXT_BASE=' + TextBase
    LFLAGS = DEVICE + ' -Wl,--gc-sections,-Map=rtthread_mini2440.map,-cref,-u,_start  -nostartfiles -T mini2440_ram.ld' + ' -Ttext ' + TextBase

    CPATH = ''
    LPATH = ''

    if BUILD == 'debug':
        CFLAGS += ' -O0 -gstabs+'
        AFLAGS += ' -gstabs+'
    else:
        CFLAGS += ' -O2'

    POST_ACTION = OBJCPY + ' -O binary $TARGET rtthread.bin\n' + SIZE + ' $TARGET \n'

elif PLATFORM == 'armcc':
    # toolchains
    CC = 'armcc'
    CXX = 'armcc'    
    AS = 'armasm'
    AR = 'armar'
    LINK = 'armlink'
    TARGET_EXT = 'axf'

    DEVICE = ' --device DARMSS9'
    CFLAGS = DEVICE + ' --apcs=interwork --diag_suppress=870'
    AFLAGS = DEVICE
    LFLAGS = DEVICE + ' --strict --info sizes --info totals --info unused --info veneers --list rtthread-mini2440.map --ro-base 0x30000000 --entry Entry_Point --first Entry_Point'

    CFLAGS += ' -I"' + EXEC_PATH + '/ARM/RV31/INC"'
    LFLAGS += ' --libpath "' + EXEC_PATH + '/ARM/RV31/LIB"'

    EXEC_PATH += '/arm/bin40/'

    if BUILD == 'debug':
        CFLAGS += ' -g -O0'
        AFLAGS += ' -g'
    else:
        CFLAGS += ' -O2'

    POST_ACTION = 'fromelf --bin $TARGET --output rtthread.bin \nfromelf -z $TARGET'

elif PLATFORM == 'iar':
    # toolchains
    CC = 'armcc'
    AS = 'armasm'
    AR = 'armar'
    LINK = 'armlink'

    CFLAGS = ''
    AFLAGS = ''
    LFLAGS = ''
`

scons 编译，提示的错误信息：

scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build/applications/application.o
In file included from /work/tools/gcc-3.4.5-glibc-2.3.6/lib/gcc/arm-linux/3.4.5/../../../../arm-linux/sys-include/signal.h:212,
                 from /work/tools/gcc-3.4.5-glibc-2.3.6/lib/gcc/arm-linux/3.4.5/../../../../arm-linux/sys-include/sys/signal.h:1,
                 from /work/002_rtt/rt-thread/include/libc/libc_signal.h:73,
                 from /work/002_rtt/rt-thread/include/rtlibc.h:35,
                 from /work/002_rtt/rt-thread/include/rtdef.h:1037,
                 from /work/002_rtt/rt-thread/include/rtthread.h:37,
                 from applications/application.c:22:
/work/tools/gcc-3.4.5-glibc-2.3.6/lib/gcc/arm-linux/3.4.5/../../../../arm-linux/sys-include/bits/siginfo.h:34: error: redefinition of `union sigval'
/work/tools/gcc-3.4.5-glibc-2.3.6/lib/gcc/arm-linux/3.4.5/../../../../arm-linux/sys-include/bits/siginfo.h:52: error: redefinition of `struct siginfo'
/work/tools/gcc-3.4.5-glibc-2.3.6/lib/gcc/arm-linux/3.4.5/../../../../arm-linux/sys-include/bits/siginfo.h:137: error: syntax error before numeric constant
/work/tools/gcc-3.4.5-glibc-2.3.6/lib/gcc/arm-linux/3.4.5/../../../../arm-linux/sys-include/bits/siginfo.h:274: error: redefinition of `struct sigevent'
In file included from /work/002_rtt/rt-thread/components/net/lwip-2.0.2/src/include/lwip/arch.h:48,
                 from /work/002_rtt/rt-thread/components/net/lwip-2.0.2/src/include/lwip/debug.h:40,
                 from /work/002_rtt/rt-thread/components/net/lwip-2.0.2/src/include/lwip/opt.h:52,
                 from /work/002_rtt/rt-thread/components/net/lwip-2.0.2/src/include/lwip/netif.h:40,
                 from /work/002_rtt/rt-thread/components/net/lwip-2.0.2/src/include/netif/ethernetif.h:4,
                 from applications/application.c:40:
/work/002_rtt/rt-thread/components/net/lwip-2.0.2/src/arch/include/arch/cc.h:58:1: warning: "ESHUTDOWN" redefined
In file included from /work/tools/gcc-3.4.5-glibc-2.3.6/lib/gcc/arm-linux/3.4.5/../../../../arm-linux/sys-include/bits/errno.h:25,
                 from /work/tools/gcc-3.4.5-glibc-2.3.6/lib/gcc/arm-linux/3.4.5/../../../../arm-linux/sys-include/errno.h:36,
                 from /work/002_rtt/rt-thread/include/libc/libc_errno.h:32,
                 from /work/002_rtt/rt-thread/include/rtlibc.h:30,
                 from /work/002_rtt/rt-thread/include/rtdef.h:1037,
                 from /work/002_rtt/rt-thread/include/rtthread.h:37,
                 from applications/application.c:22:
/work/tools/gcc-3.4.5-glibc-2.3.6/lib/gcc/arm-linux/3.4.5/../../../../arm-linux/sys-include/linux/errno.h:114:1: warning: this is the location of the previous definition
scons: *** [build/applications/application.o] Error 1
scons: building terminated because of errors.

BernardXiong, 2018-02-05:
请使用针对裸机或rtos的arm-none-eabi-gcc工具链。

closed 2018-02-05 by BernardXiong

craigtao, 2018-02-06:
前辈，但是我这个编译器也是可以编译裸机的呀，我这段时间都是用这个编译器编译裸机代码的，还有我这个编译出来的时候，是烧写到 norflash还是nandflash的？烧写地址都是0吗？
谢谢，