RT-Thread/rt-thread #1210: micropython

duanlangtaosha, 2018-02-06:
看了一下pakage中有micropython的下载链接，但是在rt-thread中怎么用呢？有没有example呢？

SummerGGift, 2018-02-06:
熟悉了在env里面下载软件包的操作之后就可以下载microPython了，目前支持gcc环境，具体的操作方法你可以把包下载下来然后查案软件包里面的readme.md 和 quickstart手册呢。

armink, 2018-02-06:
RT-Thread 的 MicroPython package 的地址在这里： https://github.com/armink-rtt-pkgs/micropython

IAR 工具链应该也是支持的

xiaoxiaoj, 2018-03-02:
如何将.py文件编译成 .pyc文件 呢

SummerGGift, 2018-03-02:
@xiaoxiaoj  可以使用microPython仓库中的mp-cross工具。
https://github.com/micropython/micropython/tree/master/mpy-cross

xiaoxiaoj, 2018-03-06:
我想在已经移植的micropython中添加pyb的LED模块，请问qstrdefs.generated.h文件中QDEF(MP_QSTR_LED,(const byte*"x88\xdc\x03" "LED"))后面的常量字符串是利用makeqstrdata.py自动生成吗

xiaoxiaoj, 2018-03-14:
该如何运行python脚本文件呢，我看移植的micropython中mpy_main()函数是可以传文件名进去的，我尝试在mpy_main.c同级目录中添加file.py，想通过串口的交互界面执行python file.py却不行，请问运行python脚本需要移植mpy-cross工具吗，我在linux下用micropython源码中的mpy-cross工具可以将py文件编译成.mpy,百度说这样只是为了保护.py文件。

SummerGGift, 2018-03-14:
@xiaoxiaoj  不必要使用 .mpy 文件，直接使用 .py 文件也是ok的。

xiaoxiaoj, 2018-03-14:
那么xxx.py文件应该放在那个路径呢，

armink, 2018-03-14:
Finsh/MSH 中输入： `python file.py` 执行脚本

xiaoxiaoj, 2018-03-14:
这样是需要挂载sd卡吗，将file.py存在sd卡中，在Finsh/MSH 中进入sd卡目录来执行file.py脚本吗

armink, 2018-03-14:
不一定的，有可能是本地有 Flash 文件系统，可以把文件传送至 Flash 文件系统

xiaoxiaoj, 2018-03-16:
@SummerGGift 
我自己写的文件可以通过python file.py来直接运行脚本了，但是尝试你说的使用Import模块名的方式来运行程序时，进入python交互界面后，import file会出错，显示找不到模块ImportError:module not found ，还需要配置什么吗？

SummerGGift, 2018-03-16:
@xiaoxiaoj  如果是你自己写的文件，你需要添加qstr，如上面所述。

xiaoxiaoj, 2018-03-16:
好的，谢谢

xiaoxiaoj, 2018-03-16:
是这个样子的，我写了两个.py文件，test1.py和test2.py，test2.py里面通过 import test1 调用test1中的函数， 在我执行python test2.py的时候，显示ImportError:module not found没有test1模块，这个test1我也要自己去在qstrdefs.generated.h里面添加文件吗，好像不需要吧，这样每一个用户写的python脚本都要自己去添加模块？

SummerGGift, 2018-03-16:
1. 你写的py文件里import了不支持的模块，这个时候需要在qstrdefs.generated.h里面添加qstr。
2. 仅仅是使用其他py里的函数，没有第一个问题的话，如果找不到我估计是你是不是放的位置不对？

xiaoxiaoj, 2018-03-16:
我写的文件test.py存在u盘中，然后u盘挂载到开发板上，进入u盘中test.py的目录，直接python test.py就可以运行，而进入python进入python交互界面后import test就不行，我已经在qstrdefs.generated.h中添加QDEF(MP_QSTR_test,(const byte*)"\x33\x04" "test")了，还有哪里不对吗*_*

closed 2021-02-09 by mysterywolf