RT-Thread/rt-thread #1292: cortex-m4的context_gcc.S文件编译报错

lixingangsms, 2018-03-16:
工具链 :arm-none-eabi 7-2017-q4-major
报错如下：
context_gcc.S:113: Error: thumb conditional instruction should be in IT block -- `vstmdbeq r1!,{d8-d15}'
context_gcc.S:122: Error: thumb conditional instruction should be in IT block -- `moveq r4,#0x01'
context_gcc.S:143: Error: thumb conditional instruction should be in IT block -- `vldmiane r1!,{d8-d15}'
context_gcc.S:155: Error: thumb conditional instruction should be in IT block -- `bicne lr,lr,#0x10'
修改如下后编译器不再报错：
``
#if defined (__VFP_FP__) && !defined(__SOFTFP__)
    TST     lr, #0x10           /* if(!EXC_RETURN[4]) */
    IT EQ
    VSTMDBEQ r1!, {d8 - d15}    /* push FPU register s16~s31 */
#endif

#if defined (__VFP_FP__) && !defined(__SOFTFP__)
    MOV     r4, #0x00           /* flag = 0 */

    TST     lr, #0x10           /* if(!EXC_RETURN[4]) */
    IT EQ
    MOVEQ   r4, #0x01           /* flag = 1 */

    STMFD   r1!, {r4}           /* push flag */
#endif

#if defined (__VFP_FP__) && !defined(__SOFTFP__)
    CMP     r3,  #0             /* if(flag_r3 != 0) */
    IT NE
    VLDMIANE  r1!, {d8 - d15}   /* pop FPU register s16~s31 */
#endif

#if defined (__VFP_FP__) && !defined(__SOFTFP__)
    ORR     lr, lr, #0x10       /* lr |=  (1 << 4), clean FPCA. */
    CMP     r3,  #0             /* if(flag_r3 != 0) */
    IT NE
    BICNE   lr, lr, #0x10       /* lr &= ~(1 << 4), set FPCA. */
#endif
``
这个问题是只有我遇到吗？还是一个BUG？

TanekLiang, 2018-03-16:
你可以用`scons --verbose`来编译， 它可以输出详细的编译命令：

```
...
arm-none-eabi-gcc -c -mcpu=cortex-m7 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -ffunction-sections -fdata-sections -x assembler-with-cpp -Wa,-mimplicit-it=thumb -o build\kernel\libcpu\arm\cortex-m7\context_gcc.o ......... \libcpu\arm\cortex-m7\context_gcc.S
...
```

然后检测一下编译参数，我觉得有可能是内核没有正确指定。

lixingangsms, 2018-03-16:
我在GCC编译参数中加上了 -v，然后最终打印出的编译命令行如下：
``
arm-none-eabi-gcc -mcpu=cortex-m4 -march=armv7e-m -mthumb -mlittle-endian -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mno-unaligned-access -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fno-common -ffreestanding -fsingle-precision-constant -flto -Wall -g3 -T "STM32F417ZG_FLASH.ld" -nostartfiles -nodefaultlibs -nostdlib -Xlinker --gc-sections -v -o …… 
``
还有，我使用的是eclipse，是在学习tr-thread源代码的时候发现这个问题的，并没有尝试用scon构建工程.难道是这个问题造成的？

TanekLiang, 2018-03-16:
问了aozima。

https://www.rt-thread.org/qa/forum.php?mod=viewthread&tid=3106

lixingangsms, 2018-03-16:
哦，百度了一下，这个选项的作用是自动插入IT指令……多谢了

closed 2018-03-16 by lixingangsms