RT-Thread/rt-thread #1295: IAR 8.22.1版本中宏_DLIB_THREAD_SUPPORT被移除

Y-JESSE, 2018-03-18:
在IAR8.22.1版本的DLib_Defaults.h文件中_DLIB_THREAD_SUPPORT被移除，并且会抛出一个错误
```
#ifdef _DLIB_THREAD_SUPPORT
  #error "Option has been removed"
#endif
```
在RT-Thread工程路径下rt-thread-master\components\libc\compilers\dlib，SConscript脚本如果检测到RT_USING_DFS宏被打开，会在IAR环境中添加宏_DLIB_THREAD_SUPPORT，最终导致编译出错

TanekLiang, 2018-03-18:
是的。IAR升级之后，这个宏被废弃了。但是旧的IAR还是依赖这个宏。
临时的解决版本是，每次生成工程后，在新版IAR里手动在选项里移除相关的宏。

后续会考虑如何合适的支持不同IAR版本。

BernardXiong, 2018-03-19:
看来有必要在scons中去检测各种已知编译器版本啊

TanekLiang, 2018-04-19:
这个问题在之前的pr已经修复了。
需要在rtconfig.py里面正确指定IAR路径，就可以在生成工程的时候，自动判断IAR路径并做处理了。

@Y-JESSE

closed 2018-05-18 by TanekLiang

skllii, 2023-02-10:
rtconfig.py已经指定IAR8.4的路径，这个编译报错还是会有
[image]

aprilhome, 2023-04-17:
这个最后怎么解决的呀，我修改了py文件，也还是不行。
[image]