RT-Thread/rt-thread #1357: 关于使用scons构建系统

sijinxiaotongxue, 2018-04-11:
能否添加一个选项使得能使用scons或menconfig工具直接在现有的工程下进行配置。现在有的方式进行重新配置后会清空工程的所有配置很不方便

BernardXiong, 2018-04-26:
具体指的是？有些不那么明白

Prismatic07, 2018-05-29:
是指裁剪功能的窗口吗？
Linux下在工程下执行`$ scons --menuconfig`就可以配置，但是要根据报错提示安装一些东西
比如要到`tools/kconfig-frontends`目录下去编译生成kconfig-mconf什么的。。。

armink, 2018-05-29:
楼主意思应该是生成的工程时，可以选择不要基于 template.uvproj ，而是基于 project.uvproj 。

这样 project.uvprojx 自己做了些改动，再次生成工程时，不会丢失吧。

BernardXiong, 2018-05-29:
如果是这样，可以对template.uvproj进行配置，这里面的一些基本配置会保留的。但是会对Group，源文件进行清空；宏定义进行追加，链接参数进行追加。

sijinxiaotongxue, 2018-07-07:
正如armink所说的，在scons对已经编辑完的工程下再添加一个新组件时，scons生成的新工程会将原有的用户修改的代码删除，变成template.uvproj的内容了。这不是很方便。建议开发者能开发出方便添加新组件而不会丢失用户代码的工具，或者提供一些文本说明不使用scons，手动添加新组件的方法。

mysterywolf, 2021-04-17:
你好，你可以修改Sconscript脚本，这样生成新工程的时候就可以保留用户的代码了。请参考：https://docs.rt-thread.org/#/development-tools/scons/scons

closed 2021-04-17 by mysterywolf