RT-Thread/rt-thread #1384: GNU Tools for Arm Embedded Processors 7-2017-q4-major error.

bestw, 2018-04-24:
bsp: stm32f7-disco
gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
config: default

error info:

scons: Reading SConscript files ...

scons: warning: No version of Visual Studio compiler found - C/C++ compilers most likely not set correctly
File "C:\Users\allbe\Documents\CWorks\stm32f7-disco\SConstruct", line 37, in <module>
scons: done reading SConscript files.
scons: Building targets ...
scons: building associated VariantDir targets: build
CC build\drivers\drv_led.o
In file included from rt-thread\include/libc/libc_signal.h:75:0,
                 from rt-thread\include/rtlibc.h:35,
                 from rt-thread\include/rtdef.h:1040,
                 from rt-thread\include/rtthread.h:37,
                 from drivers\drv_led.c:24:
c:\devtools\gcc-arm-none-eabi\arm-none-eabi\include\sys\signal.h:42:7: error: redefinition of 'union sigval'
 union sigval {
       ^~~~~~
In file included from rt-thread\include/rtlibc.h:35:0,
                 from rt-thread\include/rtdef.h:1040,
                 from rt-thread\include/rtthread.h:37,
                 from drivers\drv_led.c:24:
rt-thread\include/libc/libc_signal.h:38:7: note: originally defined here
 union sigval
       ^~~~~~
In file included from rt-thread\include/libc/libc_signal.h:75:0,
                 from rt-thread\include/rtlibc.h:35,
                 from rt-thread\include/rtdef.h:1040,
                 from rt-thread\include/rtthread.h:37,
                 from drivers\drv_led.c:24:
c:\devtools\gcc-arm-none-eabi\arm-none-eabi\include\sys\signal.h:47:8: error: redefinition of 'struct sigevent'
 struct sigevent {
        ^~~~~~~~
In file included from rt-thread\include/rtlibc.h:35:0,
                 from rt-thread\include/rtdef.h:1040,
                 from rt-thread\include/rtthread.h:37,
                 from drivers\drv_led.c:24:
rt-thread\include/libc/libc_signal.h:44:8: note: originally defined here
 struct sigevent
        ^~~~~~~~
In file included from rt-thread\include/libc/libc_signal.h:75:0,
                 from rt-thread\include/rtlibc.h:35,
                 from rt-thread\include/rtdef.h:1040,
                 from rt-thread\include/rtthread.h:37,
                 from drivers\drv_led.c:24:
c:\devtools\gcc-arm-none-eabi\arm-none-eabi\include\sys\signal.h:72:3: error: conflicting types for 'siginfo_t'
 } siginfo_t;
   ^~~~~~~~~
In file included from rt-thread\include/rtlibc.h:35:0,
                 from rt-thread\include/rtdef.h:1040,
                 from rt-thread\include/rtthread.h:37,
                 from drivers\drv_led.c:24:
rt-thread\include/libc/libc_signal.h:62:24: note: previous declaration of 'siginfo_t' was here
 typedef struct siginfo siginfo_t;
                        ^~~~~~~~~
scons: *** [build\drivers\drv_led.o] Error 1
scons: building terminated because of errors.

gcc 7中 <sys/signal.h> 和5.4比有改变，造成编译错误

BernardXiong, 2018-04-24:
目前gcc 7还未支持，c99/c11冲突了。上面的错误可以通过menuconfig配置消除，但依然还会有c99/c11的错误。

awangbrace, 2018-04-27:
我在编译RT1053的时候也出现了，当时直接屏蔽了一个头文件，这个问题就没有出现了

panzongqu, 2018-05-30:
屏蔽掉rtconfig.h的RT_USING_LIBC和RT_USING_PTHREADS，
编译通过了，我的ｇｃｃ是自己编译的，如果官方最新版本通过就做ＰＲ：

> LINK rtthread-stm32f7xx.elf
arm-none-eabi-objcopy -O binary rtthread-stm32f7xx.elf rtthread.bin
arm-none-eabi-size rtthread-stm32f7xx.elf
   text	   data	    bss	    dec	    hex	filename
  84184	   1304	   2896	  88384	  15940	rtthread-stm32f7xx.elf
scons: done building targets.

> stm32f7-disco [master●] % git rev-parse HEAD
48e769c170f5879bf3d93cc78515000f46fe4f9f

> stm32f7-disco [master●] % arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
COLLECT_LTO_WRAPPER=/home/xx/toolchain/libexec/gcc/arm-none-eabi/7.3.0/lto-wrapper
Target: arm-none-eabi
Configured with: ../gcc-7.3.0/configure --target=arm-none-eabi --prefix=/home/epanda/toolchain --with-newlib --with-build-time-tools=/home/xx/toolchain/arm-none-eabi/bin --with-sysroot=/home/xx/toolchain/arm-none-eabi --disable-shared --enable-interwork --disable-nls --enable-poison-system-directories --enable-lto --enable-gold --disable-libmudflap --disable-libgomp --disable-libstdcxx-pch --disable-libssp --disable-tls --disable-threads --disable-libunwind-exceptions --enable-checking=release --without-headers --enable-languages=c
Thread model: single
gcc version 7.3.0 (GCC)

BernardXiong, 2018-05-31:
使用github的版本依然需要屏蔽POSIX thread吗？能否提供一份cconfig.h，看看你这边的GCC配置

panzongqu, 2018-05-31:
试了[gcc-arm-none-eabi-7-2017-q4-major-linux.tar.bz2](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads)，禁掉RT_USING_LIBC和RT_USING_PTHREADS；测试commit id另外board.c 增加_init()函数，stm32f7xx_hal_conf.h增加ＮＵＬＬ的宏定义，编译是通过的．
gcc一般支持前向兼容，我们怎么调整呢？

BernardXiong, 2018-06-04:
GCC7，想等着下一版看看posix thread这部分定义是否改过来了。这样的定义挺奇怪的，他们手动改过newlib的代码。应该只要不启用pthreads就可以编译过。

所以暂时观望下，看看下一个版本的情况。否则的话，改动还不会小呢……

closed 2022-01-20 by mysterywolf