RT-Thread/rt-thread #1392: imxrt1052-evk persimmon ui  gcc编译错误

awangbrace, 2018-04-27:
使用nxp官方提供的gcc-arm-none-eabi-7-2017-q4-major．
编译基本项目没有问题．
添加gui_engine后也没有问题．
但是添加persimmon就报错．如下：
......
LINK rtthread-imxrt.elf
build/packages/persimmon/examples/images-animation/animation.o: In function `AnimationWin::AnimationWin()':
animation.cpp:(.text+0x1c): undefined reference to `Persimmon::Window::Window(char const*)'
animation.cpp:(.text+0x9c): undefined reference to `Persimmon::Rect::Rect(int, int, int, int)'
animation.cpp:(.text+0xa8): undefined reference to `Persimmon::Rect::getWidth() const'
animation.cpp:(.text+0xb8): undefined reference to `Persimmon::Rect::getHeight() const'
animation.cpp:(.text+0x148): undefined reference to `Persimmon::ImageBox::ImageBox(Persimmon::Image*, char const*)'
animation.cpp:(.text+0x1d8): undefined reference to `Persimmon::Rect::Rect(int, int, int, int)'
animation.cpp:(.text+0x230): undefined reference to `Persimmon::Widget::getRect() const'
animation.cpp:(.text+0x23c): undefined reference to `Persimmon::Rect::getWidth() const'
animation.cpp:(.text+0x25c): undefined reference to `Persimmon::Rect::getHeight() const'
animation.cpp:(.text+0x2d8): undefined reference to `Persimmon::Timer::Timer(int, Persimmon::Timer::flag)'
animation.cpp:(.text+0x31c): undefined reference to `Persimmon::Timer::start()'
build/packages/persimmon/examples/images-animation/animation.o: In function `AnimationWin::~AnimationWin()':
animation.cpp:(.text+0x380): undefined reference to `Persimmon::Window::~Window()'
build/packages/persimmon/examples/images-animation/animation.o: In function `AnimationWin::onButton()':
animation.cpp:(.text+0x428): undefined reference to `Persimmon::Animation::Animation(Persimmon::Widget&)'
animation.cpp:(.text+0x7b0): undefined reference to `Persimmon::Point::Point(int, int)'
animation.cpp:(.text+0x7d8): undefined reference to `Persimmon::Point::Point(int, int)'
animation.cpp:(.text+0x804): undefined reference to `Persimmon::AnimMoveAnimator::AnimMoveAnimator(rtgui_dc*, Persimmon::Point const&, Persimmon::Point const&)'
animation.cpp:(.text+0x818): undefined reference to `Persimmon::Widget::getBufferDrawing(Persimmon::RenderFlag)'
animation.cpp:(.text+0x884): undefined reference to `Persimmon::Animation::start(bool)'
animation.cpp:(.text+0x898): undefined reference to `Persimmon::Animation::~Animation()'
build/packages/persimmon/examples/images-animation/animation.o:(.rodata+0xdc28): undefined reference to `Persimmon::Container::setRect(Persimmon::Rect const&)'
build/packages/persimmon/examples/images-animation/animation.o:(.rodata+0xdc2c): undefined reference to `Persimmon::Container::render(rtgui_dc*, Persimmon::Point const&, Persimmon::Rect const&, Persimmon::RenderFlag)'
build/packages/persimmon/examples/images-animation/animation.o:(.rodata+0xdc30): undefined reference to `Persimmon::Widget::renderBackground(rtgui_dc*, Persimmon::Point const&, Persimmon::Rect const&, Persimmon::RenderFlag)'
build/packages/persimmon/examples/images-animation/animation.o:(.rodata+0xdc34): undefined reference to `Persimmon::Widget::onHide(rtgui_event*)'
build/packages/persimmon/examples/images-animation/animation.o:(.rodata+0xdc38): undefined reference to `Persimmon::Widget::onShow(rtgui_event*)'
build/packages/persimmon/examples/images-animation/animation.o:(.rodata+0xdc3c): undefined reference to `Persimmon::Window::eventHandler(rtgui_event*)'
build/packages/persimmon/examples/images-animation/animation.o:(.rodata+0xdc40): undefined reference to `Persimmon::Container::getMouseOwner(int, int)'
build/packages/persimmon/examples/images-animation/animation.o:(.rodata+0xdc48): undefined reference to `Persimmon::Container::addChild(Persimmon::Widget*)'
build/packages/persimmon/examples/images-animation/animation.o:(.rodata+0xdc4c): undefined reference to `Persimmon::Container::removeChild(Persimmon::Widget*)'
build/packages/persimmon/examples/images-animation/animation.o:(.rodata+0xdc50): undefined reference to `Persimmon::Window::show(int)'
build/packages/persimmon/examples/images-animation/animation.o:(.rodata+0xdc54): undefined reference to `Persimmon::Window::close(int)'
build/packages/persimmon/examples/images-animation/animation.o:(.rodata+0xdc58): undefined reference to `Persimmon::Window::dealKbd(rtgui_event_kbd*)'
build/packages/persimmon/examples/images-animation/ui_win.o: In function `ui_win':
ui_win.cpp:(.text+0x24): undefined reference to `Persimmon::Application::Application(char const*)'
build/kernel/components/cplusplus/crt_init.o: In function `cplusplus_system_init':
/home/work/work/RT-thread/rt-thread/components/cplusplus/crt_init.c:83: undefined reference to `__ctors_start__'
/home/work/work/RT-thread/rt-thread/components/cplusplus/crt_init.c:83: undefined reference to `__ctors_end__'

BernardXiong, 2018-04-27:
persimmon这个是package，应该提到persimmon那边去。persimmon只有lib，没有代码，目前也未对i.mx rt释放出匹配的库呢

closed 2018-04-27 by BernardXiong

awangbrace, 2018-04-28:
好的，明白了，多谢，现在UI除了persimmon，还有其他的选择吗？